pascallanger/DIY-Multiprotocol-TX-Module #169: Can’t get the Additional Boards Manager URLs to work in Arduino IDE.

flyrcpro, 2018-06-04:
I type in “https://raw.githubusercontent.com/pascallanger/DIY-Multiprotocol-TX-Module/master/BootLoaders/package_multi_4in1_board_index.json” in the Arduino IDE Preferences at Additional Boards Manager URLs.
But the Multi 4 in 1 boards do not show up in the Boards Manger.
How do I get the Multi 4 in 1 boards to show up under Boards Manager?
My TX is a Turnigy 9RX Pro with ersky9x r220 installed.
I want to flash the latest firmware to my iRangeX IRX4 Plus.  The iRangeX IRX4 Plus has the STM32 Board.  Because the Multi 4-in-1 STM32 Board does not show up in Boards Manager, I get nowhere.
So far I have loaded the bootloader with a FTDI, but now my iRangeX IRX4 Plus is bricked.
The key is to get the 4 in 1 boards to show up under Boards Manager.
Any help would be appreciated.
Gary

per1234, 2018-06-05:
Which version of the Arduino IDE are you using?

pascallanger, 2018-06-05:
The URL is wrong, if you try to click on your own url you'll see that you get an error 404 not found instead of getting the json file.
Do a copy and paste from the instructions.

Also use the forum or GitHub but not both to get help...

Pascal

closed 2018-06-05 by pascallanger

flyrcpro, 2018-06-05:
Thanks for the quick response per1234.  I’m using Arduino 1.8.5 for windows on a windows 10 computer.