pascallanger/DIY-Multiprotocol-TX-Module #171: Documentation on current Banggood modules

andrejpodzimek, 2018-06-21:
With my [Banggood 4-in-1](https://www.banggood.com/CC2500-NRF24L01-A7105-CYRF6936-4-In-1-RF-Module-For-Walkera-D7E-D10-Transmitter-p-1096707.html) module I’ve recently had a [telemetry issue](https://github.com/betaflight/betaflight/issues/6145) when using the latest FrSky RX firmware (for R-XSR on F.Port). I got a recommendation to update the TX module firmware. However, the howtos in this repository describe modules that look quite different. I assume, based on the pinout mentioned in the product description, that what I have might be the SPI variant with ATmega328p and it has 6 undocumented solder pads on it that look like a place to attach the 6-pin header. But is it actually compatible with this firmware? Or is it a completely different device? If it’s compatible, would it please be possible to list / document it somewhere?

tipouic, 2018-06-21:
Hi,
This is not a multiprotocol module but it is a module that groups the 4 transmitters in the same box, so for your code problem, it is necessary to refer to that of your radio.

andrejpodzimek, 2018-06-21:
I see. Thanks for the clarification. So in that case I’ll file a bug with Deviation, the radio firmware. At least now I understand why it has 4 chip select wires. There’s no MCU, so the TX uses chip select to make (only) one of the chips work at a time and talks to it directly.

closed 2018-06-21 by andrejpodzimek

andrejpodzimek, 2018-06-21:
[Filed.](https://github.com/DeviationTX/deviation/issues/415)