pascallanger/DIY-Multiprotocol-TX-Module #172: Bootloader mode IRX4Plus

pinkywafer, 2018-06-21:
This issue relates to the documentation for flashing the bootloader to the Irange IRX4plus.

The documentation shows the Boot0 and 3v3 pins on the bottom row of pins being jumped to enter bootloader mode, but on my version of the module, it would not connect with jumper in this position - giving a 'Failed to init device' message.
Following a search on RC Groups, it appears that others have had this issue, and that now the correct way to enter bootloader mode is to short the pins on both rows above Boot0.  I tried this method, and it worked perfectly.
I am not sure if there was an earlier version that used the method shown in the documentation, but if so, it seems that the current version has this method.  
I did try to photograph this, but as I only have black jumpers, it is not clear on the photo.

pascallanger, 2018-07-20:
I think I've captured it in the doc. Please let me know if not...

closed 2018-07-20 by pascallanger

pinkywafer, 2018-07-20:
Not quite, you don't need 2 ...

If it doesn't work move the jumper to bridge the two left hand pins (BOOT0 and directly above).

Would be better

pascallanger, 2018-07-20:
updated