pascallanger/DIY-Multiprotocol-TX-Module #174: Battery Voltage Telemetry AFDS2A

maz3max, 2018-07-17:
Hi!
I get a constant value for the battery voltage of my Turnigy iA6C RX.
On a Turnigy i6s radio with Turnigy firmware, I get two voltages as telemetry: the 5V and battery voltage.
Could the value be the steady 5V value?
Can I somehow log the unknown sensor ids found in AFHDS2A_a7105.ino:103?
Maybe it's just a different id?

maz3max, 2018-07-17:
I use a Turnigy 9XR with er9x.

maz3max, 2018-07-17:
They defined 0x03 as external voltage. This might be exactly what I need. How can I map this to A1 or A2?

pascallanger, 2018-07-17:
er9x does not support full AFHDS2A telmetry like ersky9x or OpenTX does. The output is limited to what FrskyD allows to send to the radio.

pascallanger, 2018-07-17:
You could try something like this in the multi code in AFHDS2A_a7105.ino with no warranty:
#if defined(AFHDS2A_FW_TELEMETRY) || defined(AFHDS2A_HUB_TELEMETRY)
// telemetry sensors ID
enum{
	AFHDS2A_SENSOR_RX_VOLTAGE   = 0x00,
	AFHDS2A_SENSOR_RX_ERR_RATE  = 0xfe,
	AFHDS2A_SENSOR_RX_RSSI      = 0xfc,
	AFHDS2A_SENSOR_RX_NOISE     = 0xfb,
	AFHDS2A_SENSOR_RX_SNR       = 0xfa,
	AFHDS2A_SENSOR_A3_VOLTAGE   = 0x03,
};

static void AFHDS2A_update_telemetry()
{
	// AA | TXID | rx_id | sensor id | sensor # | value 16 bit big endian | sensor id ......
	// max 7 sensors per packet
#ifdef AFHDS2A_FW_TELEMETRY
    if (option & 0x80)
	{
        // forward telemetry to TX, skip rx and tx id to save space
        pkt[0]= TX_RSSI;
        for(int i=9;i < AFHDS2A_RXPACKET_SIZE; i++)
            pkt[i-8]=packet[i];

        telemetry_link=2;
        return;
    }
#endif
#ifdef AFHDS2A_HUB_TELEMETRY
	for(uint8_t sensor=0; sensor<7; sensor++)
	{
        // Send FrSkyD telemetry to TX
		uint8_t index = 9+(4*sensor);
		switch(packet[index])
		{
			case AFHDS2A_SENSOR_RX_VOLTAGE:
				//v_lipo1 = packet[index+3]<<8 | packet[index+2];
				v_lipo1 = packet[index+2];
				telemetry_link=1;
				break;
			case AFHDS2A_SENSOR_A3_VOLTAGE:
				//v_lipo2 = packet[index+3]<<8 | packet[index+2];
				v_lipo2 = packet[index+2];
				telemetry_link=1;
				break;
			case AFHDS2A_SENSOR_RX_ERR_RATE:
				RX_LQI=packet[index+2];
				break;
			case AFHDS2A_SENSOR_RX_RSSI:
				RX_RSSI = -packet[index+2];
				break;
			case 0xff:
				return;
			/*default:
				// unknown sensor ID
				break;*/
		}
	}
#endif
}
#endif

maz3max, 2018-07-17:
I got it to work, but there was another problem:
The voltage values are two bytes long. I had to squash those two bytes into one.
The changes in this commit work:
https://github.com/maz3max/DIY-Multiprotocol-TX-Module/commit/942f6bb2e89888989927f96d45fc4a860556d063

pascallanger, 2018-07-17:
You really drop in your code the low part of the voltage... Do you really need to divide by 8 (>>3)?

maz3max, 2018-07-17:
I did some empirical testing with 4S batteries and this is the best setting I found. It should work for voltages up to about 21V.

pascallanger, 2018-07-18:
with a divide by 8?

maz3max, 2018-07-18:
Yes..

pascallanger, 2018-07-20:
Will be by default in the next push.

closed 2018-07-20 by pascallanger