pascallanger/DIY-Multiprotocol-TX-Module #175: Serial mode on Turnigy 9xr Pro

QUERB, 2018-07-28:
Greetings,

I have fitted my transmitter turnigy 9xr pro with a banggood module 4 in 1 Atmega328p.
I use PPM mode without any issue for many months.
However, I would like to use now the serial mode between the transmitter and the module.
So, I have upgraded the firmware of the transmitter with success (version 8r222 of june 2018).
In addition, I have solded the Tx and the Rx pads of the module.
Now, if I select the 0 position for serial mode on the module and select multiprotocole (type AFHDS2A for example) on protocole menu of the transmitter for the RC model, the red led of the module flash regurlarly with a frequency equal to the "Slow blink (on=0.5s, off=0.5s)". So the issue is that no valid signal is being seen on the Rx pin as described on the module troubleshooting manual.
At the present time, I don't know what I have to do to solve this issue.
May you help me to solve this issue ?
Thank you very much !

pascallanger, 2018-07-28:
I'm thinking you haven't soledered correctly the pads or done a shortcut between the two.
If you look carefully at the picture: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/images/V2b_ISP.jpeg you'll see that the 4 pads are not connected together but vertically (red line) they must not touch each others.

Pascal

QUERB, 2018-07-28:
Thank you for the picture.
I have effectively soldered the 2 pads of Tx and the two pads of Rx separately.
Then, I have checked the soldering : there is no contact between Tx and Rx (no electric continuity).
Perhaps, the quality of the soldering is not correct ?
Please, look at this picture 
[image]
Is it correct ?

Bruno

pascallanger, 2018-07-29:
You might have a really old version of the avr module...
On the old ones you had to do this [modification](https://camo.githubusercontent.com/ba9d480a539e25ce895b3357c46eaf4ec3e4d9eb/687474703a2f2f7374617469632e726367726f7570732e6e65742f666f72756d732f6174746163686d656e74732f342f302f382f352f382f332f61383839353033382d3137302d34696e312532306d6f64756c652e6a7067).

pascallanger, 2018-07-29:
You should check the connectivity between the RX pin of the atmega and the PPM pin of the connector. This is all what is needed to get serial working.

Pascal

QUERB, 2018-07-29:
I have bought the module just one year ago.
Ok, thank you for the details of the improvement. As soon as I have done the modification, I contact you back.

Bruno

QUERB, 2018-08-02:
I have done the modification on the module as described. The connectivity between the RX pin and the PPM pin is correct. Unfortunately, it doesn't work : the red led flashes as Slow blink. Which other improvement is possible ?
Thank you.

pascallanger, 2018-08-02:
There is no othe improvements... Like I said before you need to test the continuity from the socket PPM pin to the atmega328p RX pin.

QUERB, 2018-08-02:
There is effectively a resistor value of 2.3 kOhms.

pascallanger, 2018-08-09:
Are you sure you are configuring correctly the radio?
Is it ersky9x right?
The 9XR Pro has the capability to have 2 modules one internal and one external, you have selected and configured the external right and internal off?
You said in your first post that on position 0 the LED blinks to indicate no serial signal, what if you switch to 1 (restart the radio), is the module LED flash indicating no PPM signal as well?

QUERB, 2018-08-09:
I'm actually on holidays ; I will test it the next week and contact you back. Thank you.

pascallanger, 2018-08-21:
No comment since 12 days, I'm condidering it's working.

closed 2018-08-21 by pascallanger