pascallanger/DIY-Multiprotocol-TX-Module #176: STM compilation fails on Linux

z3r0c00l2005, 2018-08-04:
Compiling on a Linux system using multi 4in1 for STM targets fails.

The tools/linux/nnnn scripts have been saved with MSDOS format line feeds.  running the scripts through dos2unix removes the erroneous line feed characters and compilation then works OK

pascallanger, 2018-08-04:
Have you checked the compilation guide: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#linux-64-bit
There is a whole section about Linux. If this is not there, could you propose a pull request?

benlye, 2018-08-04:
When I built the previous boards archives I always did it on a Linux system to avoid this problem. Want me to rebuild the current ones and update them? I have a script to automate it that I'll push as well.

pascallanger, 2018-08-04:
Thanks Ben, only the STM version needs to be redone. This is the one that I've touched to use the latest bootloader.

pascallanger, 2018-08-09:
@z3r0c00l2005 Ben has updated the package. Please update your STM 1.0.6 board (uninstall/install) and confirm that everything works fine.

benlye, 2018-08-09:
I just pushed #179.  Rather than create a new version I just rebuilt the package for v1.0.6.

@z3r0c00l2005, if you remove and then re-add the Multi STM32 board package using the Arduino IDE Boards Manager, you should see the execute bit set correctly on all the necessary Linux files.

closed 2018-08-10 by pascallanger