pascallanger/DIY-Multiprotocol-TX-Module #177: Unable to reflash USB bootloader

z3r0c00l2005, 2018-08-06:
I'll start by saying this could just be me....but it was frustrating me all of yesterday!

I have an IRX4+ and originally flashed the USB bootloader to it, and was successfully upgrading over USB.

Then I flashed a copy using Serial - This disabled the USB bootloader for me.

Upon reflashing the bootloader, the module is detected as Maple DFU (or Maple 003) but is not then accessbile under the "ports" menu in the Arduino IDE.

This is happening on both a Linux & Windows environment.

I am, however, still able to upgrade the module over Serial connection so all is not lost, it's just not quite as user friendly of course!

Any thoughts of what to try are much appreciated.

pascallanger, 2018-08-06:
This is quite simple...
When the module is recognized as a DFU, unplug the module from USB, launch a USB upload from Arduino, wait for the stm32 uploader to launch and say something like "Searching for DFU", plug the module USB and it will be flashed.
If you think of a way to improve the doc, do you mind to do a pull request?

z3r0c00l2005, 2018-08-06:
I may have a different problem....I can't even start an upload as it says I need to choose a port first.

pascallanger, 2018-08-06:
And you don't have any COM port to select, even something not related to multi at all?

z3r0c00l2005, 2018-08-06:
Correct, nothing listed.

I'm happy this is probably just user error and since serial upload works OK it's not the end of the world.  I've probably corrupted something on the module rather than there being a code issue!

pascallanger, 2018-08-06:
No, no, just plug anything that would create a com port on the PC, don't care what it is so Arduino sees it, select it, then unplug, flash firmware, wait for it to search for DFU and plug the module.

z3r0c00l2005, 2018-08-07:
Hi Pascal, Many thanks for your help on this one - back up and running on USB now.

There's nothing wrong with the instructions in the manual........just me not reading them!!!!!

closed 2018-08-07 by z3r0c00l2005