pascallanger/DIY-Multiprotocol-TX-Module #181: 15 limit

tj51, 2018-08-11:
there is a limit for receiver number and the limit is 15, not enough, please double or quadruple the number

pascallanger, 2018-08-12:
Do you really have more than 16 (0-15) RXs of the same protocol?

pascallanger, 2018-08-21:
No comment since 10 days, I'm condidering it's ok as it is.

closed 2018-08-21 by pascallanger

Gfyss, 2019-05-28:
My apologies for the necro, but I'm relatively new to the OpenTX/multi protocol world and am playing catch up.  Please let me know if I should create a new issue instead.

Yes, I'm definitely over the 16 rx limit in the DSM2/DSMX world.  I can only use 16 of 64 slots on my X9E, I still have 4 aircraft on the old radio, and another 5 rx's for projects on hold, all of them are DSM2/DSMX flavors.  I did buy 2 FrSky rx's as a backstop, but prefer DSM's use of satellites for path diversity.

Assuming you are willing to make the adjustment and it's technically workable, my recommendation would be to increase it to the maximum number of slots allowed by OpenTX.  My X9E allows 64.  That would pretty much accommodate any future growth requirements for a long time.