pascallanger/DIY-Multiprotocol-TX-Module #182: No binding to esky F150 v2

labomat, 2018-08-12:
Trying to bind a Esky 150 micro heli v2 ("Airwolf") without success to a Taranis with iRangeX module.

Module is in serial mode (both red and green leds are solid).

Model settings:
External RF: Multi - Esky150
Version: v1.2.0.17
Tried the bind option on the tx.

On heli there is a blue led binking and after a few seconds a green led flashing.

On the deviation forum there is a discussion thread about different versions of the protocoll. Is the latest version used here?
https://www.deviationtx.com/forum/6-general-discussions/6446-esky-150x-which-protocol?start=80

pascallanger, 2018-08-12:
The protocol worksand is up to date. It's just that this little heli is not so easy to bind.
You need to bind at the right time. I can't remember exactly when but try at different time.
Also you have to do a specific stick movement to start the heli. It's in the doc of the heli.

pascallanger, 2018-08-21:
No comment since 9 days, I'm condidering it's working.

labomat, 2018-08-29:
No, unfortunately it's not. I tried every possible moment with no luck. 
I also wrote to rcgroups where a guy could reproduce the problem: https://www.rcgroups.com/forums/showthread.php?3050896-Esky-f150-v2-can-bind-to-taranis-transmitter

pete9999, 2020-04-13:
Hi,
I also have this issue and cannot get the esky 150 v2 to bind...
I have a t16 pro v2 with internal multi module. tx is running 1.3.0.52 RETA for multi module
setup a new model in opentx
inputs to 
ch1 Rud
ch2 Ele
ch3 Thr
ch4 Ail

mixer to same

tried using 4 ch, or 7 ch in multi setup to no avail.
tried adding in ch5,6,7 normally and then setting to zero as some have suggested.

Nothing will allow me to bind, Just get the flashing blue and green lights on heli..

Once the green and blue lights are flashing you can bind with the original transmitter so that appear to be the right time to bind, but the multi module wont.

Is this a problem with this protocol? or Can anyone suggest any other things to try or share a opentx model that works?

Thanks in advance

pascallanger, 2020-04-13:
It's a long time ago (couple of years) but I remember that the esky150 airwolf is a pain to bind. The bind with the original TX works each time because the heli already knows it...
You need to do the bind at the right time. If I remember correctly you have to power it on, wait for the LED pattern to change then only bind but if you wait too long it gives up, so it's all about timing. You also need to do something with the sticks to activate it which is indicated in the documentation if I remember.
I have a friend which has one (the one I used to reverse the protocol) but with the confinment I don't know when I'll be able to borrow it.
Pascal

pascallanger, 2020-04-13:
Just realized, could it be another version? The one I've worked on was esky150x airwolf.
If yes I need someone to send me the TX and heli so I can see what has changed and reverse the protocol if possible.
Pascal

labomat, 2020-04-14:
I think they made a new version some time ago -> v2

There are two major upgrades from ESKY F150X and it seems they changed something at the rc part:

1. Transmitter Upgrade
- Redesign the Tx appearance
- More reliable 2.4GHz Tech, not easy to interrupt
- Transmitter support up to 18 channel for future use
(F150 V2 just used 5 channel)
2. Structure Upgrade
- Tail Vertical Fin LED Light design
- Rotor Head upgraded that let the flight be more smooth
- Pushrod upgraded that let the control be more accurate
- CC3D Flight control upgraded that let the helicopter be more stable

pascallanger, 2020-04-14:
Without the remote and heli there is nothing I can do... If someone is willing to ship it to me I will have a look.

labomat, 2020-04-14:
I have the heli - is there something I can do from here? Taking some measurements?

pascallanger, 2020-04-14:
Do you a logic analyzer that you can connect on the SPI bus?
Can you open the TX to see what RF chip is being used (picture of the rf board)?

labomat, 2020-04-14:
The main chip is labeled (as far as I can read it):
N5 1822
QFAAH1
16 (or "0") 21CX

And yes, I have one of theses cheap usb logic analyzers - but not used it yet ... So I would giv it a try this evening if you can tell me what to measure.

pascallanger, 2020-04-14:
Can you do some pictures? Is the RF going out direct from this main chip?

labomat, 2020-04-14:
[image]
[image]
[image]

pascallanger, 2020-04-14:
Oh that's an arm + 2.4GHz in a single chip... So nothing to connect to for analyzing.
SDR is the only chance at this stage...
Can you see the receiver side components?

labomat, 2020-04-14:
[image]

labomat, 2020-04-14:
[image]

schwabe, 2020-04-14:
That is just a STM32 and a [flash](https://www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/m25p/m25p16.pdf)

The rf chip is not in the picture. I might be on another board or the backside of that board.

labomat, 2020-04-14:
[image]

labomat, 2020-04-14:
Looks like the same as on the tx ...

pascallanger, 2020-04-14:
SDR is the only way at this stage...

pascallanger, 2020-04-14:
> That is just a STM32 and a [flash](https://www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/m25p/m25p16.pdf)
> 
> The rf chip is not in the picture. I might be on another board or the backside of that board.

Welcome back! Long time not seeing you around.

labomat, 2020-04-14:
So I guess it is not possible to get some information by analyzing the signaling between rf module and the rest of the system. I do not have any experience with sdr and do not have any equipment. 
I ve seen some cheap sdr usb sticks on amazon - are they suitable for recording the communication between tx and rx?

schwabe, 2020-04-14:
@pascallanger I am still lurking and reading the mail from github but almost no time to really work on OpenTX/Multi :( The curiosity what RF would be on that board was strong enough to peek into this issue

labomat, 2020-04-14:
I asked the manufactorer about the protokoll - this is what he told me:

> Halo my friend, thank you for your support our product, for the old version protocol, we are using the fixed 2.4 frequency, so we heard many user can use their own receiver, but our F150V2 are using non-fixed frequency to avoid the lose control problem, the frequency will change every seconds, we are sorry to let you know we cannot provide you.
Does this mean it is generally impossible to use the 4in1 module for this heli?

pascallanger, 2020-04-15:
All protocols do frequency hopping... Their old protocol was using only 2 channels. FrSky is using 47 channels. So what they are saying is more that they don't want to give you the protocol they are using. I've never seen a company doing so anyway.
I'm using this SDR: https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/adalm-pluto.html#eb-overview

labomat, 2020-04-15:
is it possible to send private messages here - in order to exchange adress data?

pascallanger, 2020-04-15:
Contact me by MP on RCGroups (hpnuts): https://www.rcgroups.com/forums/showthread.php?t=2165676&goto=newpost

pascallanger, 2020-04-16:
If you can't contact me on RCG, look at my GitHub's profile, my email is there. Or press the donate button, my email address is there as well.

labomat, 2020-04-28:
Another one unsuccessfully trying to bind the 150 v2 (different model but the same internal parts): https://www.youtube.com/watch?v=XmUIdMsBnZI&feature=youtu.be

pascallanger, 2020-04-28:
I'm not sure what you are looking me to do...
I've proposed you to ship me the heli+TX or I purchase the heli with a participation from your side. I haven't received an answer since a week or two.
There is nothing I can do on this protocol remotely which we know will be hard to reverse anyway.
Pascal

labomat, 2020-04-28:
Just stumbled upon this video - messaged you on rc groups.

pascallanger, 2020-05-05:
I've just received the heli. I haven't tried it yet but instead powered on the TX. I can see using SDR that the TX is transmitting on a large number of channels which we don't care at this stage. But the trouble I have is that I can't find any GFSK information in the signal. The 51822 MCU on the picture supports the 3 NRF24L modes (RF chip which the multi module has) and 2 bluetooth low energy modes. The bad news is that the signal I'm seeing does not match at all what a NRF sends where basically the phase is linear instead of jumping based on bit values.
I'm quite puzzled at this stage... I need to power on the heli to see if the TX is behaving differently but I'm not putting a lot of hope in it...

pascallanger, 2020-05-06:
Good news!!! I've finally figuerd it out!!!
They are running in NRF24L01 compatibility mode at 250Kb/s.
Here is just a preliminary analysis:
 - TX address: FA 87 5B 2C 5D
 - 3 unknown bytes: 41 08 00
 - 5 channels coded on 2 bytes: THR, RUD, ?, ?, ?. Range from 200=0x0C8 to 1000=0x3E8 and centered on 600=0x258
 - 26 bytes set at 0x00 -> could be 13 additional channels (18 in total)
 - CRC 2 bytes: I haven't checked but I assume it's the standard CRC
There does not seem to have any extra checks in the payload.

Left to do:
 - channel assignment: trivial
 - hopping frequencies: not so easy since it hops a lot but doable
 - catch a bind packet and analyse the bind sequence -> the TX does not stay long in bind mode which is going to make the task a bit difficult
Pascal

pascallanger, 2020-05-06:
The bind is out of norm... It is sent on channel 0 at 250Kbps but with 142 bytes payload... This is something a NRF can't do.
The bind contains:
 - Bind address: 73 73 74 63
 - 142 bytes payload!
    - repeat of bind address: 73 73 74 63
    - TX address: 87 5B 2C 5D
    - 3 unknown bytes: 41 88 41
    - 4 * 0xAA
    - 110 unknown values -> hop table??
    - 25 * 0x00
  - CRC 2 bytes

The CC2500 should be able to send all this data... But wow that's a lot of data in a single packet to decrypt...

Pascal

pascallanger, 2020-05-07:
Ok I can fly the heli using the multi module configured to use the original TX ID and hopping frequencies which the heli is already bound to.
Now the problem is to send the bind frame...

pascallanger, 2020-05-08:
Protocol fully reversed. It is available if you compile from master. I will do a release a bit later.

closed 2020-05-08 by pascallanger

pascallanger, 2020-05-08:
Release: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.0.95
Please test and report.
Pascal

labomat, 2020-05-09:
@pascallanger Binding and flying seem to work flawless! Superb work!

yakacm, 2024-08-11:
I've got this heli, but mine is the V3 version, I have tried the Esky V2 protocol on my TX16S MkII, Max, with built in 4-in-1, but it doesn't map the functions correctly, I can't get the throttle  to work whatever I try. Any ideas I could try?

pitts-mo, 2024-08-11:
I am not sure about your specific instance. However, throttle is often
restrained by channel position, calibration, or an Arming switch position.

yakacm, 2024-08-11:
Hey thanks for getting back to me, I thought about the arming switch, and I don't see anything, and moving all the switches has no effect. "throttle is oftenrestrained by channel position, calibration" could you elaborate on that, not too sure what you mean by, channel position or calibration.Thanks Andy.

pitts-mo, 2024-08-11:
https://www.esky-rc.com/Uploads/202108/ESKY150V3_Manual.pdf

Suggest investigate throttle cut and the troubleshooting sections of the
manual.

Maybe see if you can find a youtube video that covers configuration of the
F150 (even older versions) in opentx/edgetx.

pascallanger, 2024-08-12:
150v3 flies perfectly fine. A friend as one and i've been able to configure it. The channels are not as described in my GitHub since for whatever reason they changed the order. I recommend removing the blades while playing with the config since his heli took off unexpectedly while I was figuring out the details. 
I think I still have the model on my radio. If that's the case I'll post the quick config I did at the time.

yakacm, 2024-08-13:
Right oh guys, my E150 is now flying. I started again and created a new model, looking at protocol details up here on Github, where it mentions that Ch5 is rate, so I set a switch for channel 5, and then just took a guess at Ch6 being throttle cut, and that did it. I was only able to fly for about 20 seconds, before the batter needed charging, just because I'd spent so long trying to trouble shoot. Just charging the battery now, I'll need to buys some more. Anyway thanks Pascal and pitts-mo for your help, I didn't think I'd ever get this thing flying.