pascallanger/DIY-Multiprotocol-TX-Module #183: image not found

fabioeh, 2018-08-14:
Getting this error when uploading with usb to MPM module .

An error occurred while uploading the sketch
  Referenced from:  /Users/fabioehrengruber/Library/Arduino15/packages/multi4in1/hardware/STM32F1/1.0.7/tools/macosx/dfu-util/dfu-util
  Reason: image not found

fabioeh, 2018-08-14:
Complete eror messages.

Sketch uses 74828 bytes (57%) of program storage space. Maximum is 131072 bytes.
Global variables use 4168 bytes (20%) of dynamic memory, leaving 16312 bytes for local variables. Maximum is 20480 bytes.
dyld: Library not loaded: /opt/local/lib/libusb-1.0.0.dylib
An error occurred while uploading the sketch
  Referenced from: /Users/fabioehrengruber/Library/Arduino15/packages/multi4in1/hardware/STM32F1/1.0.7/tools/macosx/dfu-util/dfu-util
  Reason: image not found
/Users/fabioehrengruber/Library/Arduino15/packages/multi4in1/hardware/STM32F1/1.0.7/tools/macosx/maple_upload: line 53: 19262 Abort trap: 6           ${DFU_UTIL} -d ${usbID} -a ${altID} -D ${binfile} -R ${dfuse_addr} -R

benlye, 2018-08-14:
Can you enable verbose messages on upload and repost the entire output? Also, what version of Mac OS are you on?

benlye, 2018-08-14:
I was able to reproduce this on my test Mac (running High Sierra).  It looks like the tools we use to write to the module, dfu-util, has an unmet  dependency on libusb.

The quick solution was to install [Homebrew](https://brew.sh/) then install libusb:

```
/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
brew install libusb
```

The part I don't understand is I'm *sure* I've uploaded via USB on OS X before without this problem.  The only other thing I've changed recently is installing the Command Line Developer Tools.

It looks like we'll have to update the docs for Mac.

fabioeh, 2018-08-14:
Last login: Tue Aug 14 07:15:19 on console
Fabio-MacBook-16:~ fabioehrengruber$ ls -al /usr/local/lib
total 0
drwxrwxr-x   2 fabioehrengruber  admin   64 Aug 14 07:04 .
drwxr-xr-x  15 root              wheel  480 Aug 14 07:04 ..
Fabio-MacBook-16:~ fabioehrengruber$

fabioeh, 2018-08-14:
Fabio-MacBook-16:~ fabioehrengruber$ /usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
==> This script will install:
/usr/local/bin/brew
/usr/local/share/doc/homebrew
/usr/local/share/man/man1/brew.1
/usr/local/share/zsh/site-functions/_brew
/usr/local/etc/bash_completion.d/brew
/usr/local/Homebrew

Press RETURN to continue or any other key to abort
==> /usr/bin/sudo /bin/mkdir -p /Library/Caches/Homebrew
Password:
==> /usr/bin/sudo /bin/chmod g+rwx /Library/Caches/Homebrew
==> /usr/bin/sudo /usr/sbin/chown fabioehrengruber /Library/Caches/Homebrew
==> Downloading and installing Homebrew...
HEAD is now at 3e890c11a Merge pull request #4679 from sjackman/os-linux
Updated 1 tap (homebrew/core).
==> Cleaning up /Library/Caches/Homebrew...
==> Migrating /Library/Caches/Homebrew to /Users/fabioehrengruber/Library/Caches
==> Deleting /Library/Caches/Homebrew...
==> Updated Formulae
openssl@1.1
==> Installation successful!

==> Homebrew has enabled anonymous aggregate user behaviour analytics.
Read the analytics documentation (and how to opt-out) here:
  https://docs.brew.sh/Analytics.html

==> Next steps:
- Run `brew help` to get started
- Further documentation: 
    https://docs.brew.sh
Fabio-MacBook-16:~ fabioehrengruber$ ls -al /usr/local/lib
total 0
drwxrwxr-x   2 fabioehrengruber  admin   64 Aug 14 07:04 .
drwxr-xr-x  15 root              wheel  4Fabio-MFabio-FabioFabiFaFabiFabFabiFFFaFabFabFFaFFaFabiFaFaFabio-MFFabio-MFabFabio-Fabio-MaFabFaFFabio-MacBook-16:~ fabioehrengruber$

benlye, 2018-08-14:
So I think that means that libusb is still not actually installed, or is not linked correctly.

In the output above you have only installed brew.  Did you run the second command, after installing brew?
`brew install libusb`

fabioeh, 2018-08-14:
working now thanks

Last login: Tue Aug 14 08:58:10 on ttys000
Fabio-MacBook-16:~ fabioehrengruber$ brew install libusb
==> Downloading https://homebrew.bintray.com/bottles/libusb-1.0.22.high_sierra.b
######################################################################## 100.0%
==> Pouring libusb-1.0.22.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/libusb/1.0.22: 29 files, 514.8KB
Fabio-MacBook-16:~ fabioehrengruber$ ls -al /usr/local/lib
total 0
drwxrwxr-x   6 fabioehrengruber  admin  192 Aug 14 09:13 .
drwxr-xr-x  15 root              wheel  480 Aug 14 07:04 ..
lrwxr-xr-x   1 fabioehrengruber  admin   46 Aug 14 09:13 libusb-1.0.0.dylib -> ../Cellar/libusb/1.0.22/lib/libusb-1.0.0.dylib
lrwxr-xr-x   1 fabioehrengruber  admin   40 Aug 14 09:13 libusb-1.0.a -> ../Cellar/libusb/1.0.22/lib/libusb-1.0.a
lrwxr-xr-x   1 fabioehrengruber  admin   44 Aug 14 09:13 libusb-1.0.dylib -> ../Cellar/libusb/1.0.22/lib/libusb-1.0.dylib
drwxr-xr-x   3 fabioehrengruber  admin   96 Aug 14 09:13 pkgconfig
Fabio-MacBook-16:~ fabioehrengruber$

benlye, 2018-08-14:
Excellent.  I've updated the [doc](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#mac-os-x).

closed 2018-08-14 by benlye