pascallanger/DIY-Multiprotocol-TX-Module #187: Update MPM Lite

amfiber, 2018-08-22:
Can someone Please tell me A easy way to update this module. I read the general part but I worry to brick the module

pascallanger, 2018-08-22:
First, you can't brick a module as there is always the option to start in Boot0.
Second, the MPM Lite module already comes with the bootloader installed so you just need to plug it and upload the new firmware. All this is explained in details here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#compiling-and-programming-stm32

Dimonira, 2018-08-23:
You also need to flash the X-Lite transmitter firmware with multimodule build options.

amfiber, 2018-08-23:
Will try thanks

closed 2018-08-23 by pascallanger

amfiber, 2018-08-24:
I cannot get it to pop-up by com port
Only com 6

pascallanger, 2018-08-24:
Have you installed the drivers?

amfiber, 2018-08-24:
Yes

pascallanger, 2018-08-24:
So what do you see in the device manager when you plug the module?

amfiber, 2018-08-24:
Sorry Windows was restarting
Serial USB com 6

pascallanger, 2018-08-24:
So you are all good just upload.

amfiber, 2018-08-24:
Error you must select One of these boards

pascallanger, 2018-08-24:
The one given in the documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Arduino_IDE_Boards.md

amfiber, 2018-08-24:
Now I see com
Com 6 multi 4-1 stm32..... 👍🏻

pascallanger, 2018-08-24:
Please follow the doc line by line and you will see that it works easily:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#compiling-and-programming-stm32

amfiber, 2018-08-24:
Yessss. Missed a step😬😗
But it works now 👍🏻
Can I make a small donation with Paypal?

pascallanger, 2018-08-24:
Always welcome :-)
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Donations.md

amfiber, 2018-08-24:
Keep up the good work and many thanks for helping me out