pascallanger/DIY-Multiprotocol-TX-Module #193: [Enhancement request] Multi & OpenTX (serial mode) sending protocol information

LapinFou, 2018-09-23:
Hi,

Again, thanks a lot for wonderful job. 👍 
I posted a comment in the OpenTX GitHub → https://github.com/opentx/opentx/pull/6183
The idea is to have a text file on the SD Card will all protocols/sub-protcols name. Thus, if you add something in the Multi, we don't need to wait the new OTX version.
@schwabe made a good point. Wouldn't be simpler if the multi send the protocol information?

I'm not a soft guy. This would probably require lot of work. However for the end user it would much simpler.

What do you think?

pascallanger, 2018-09-27:
I'll look at it later since I've got too much on my plate right now.

LapinFou, 2018-09-27:
I do understand. 😃 
What is your opinion on this idea?
Do you think it can be done?

pascallanger, 2018-09-27:
Yes because it makes sense but I would like it compatible with both ersky9x and OpenTX which will take sometime to get the devs all agree on the implementation way. I need to first give it a go to lay down a proposition which is what I'm pushing to later for now.

LapinFou, 2018-09-27:
The most important is that you think it is a good idea.
Hopefully it will be a 2019 new feature. 😉

raphaelcoeffic, 2019-09-03:
Or 2020? 😁

Ideally, the list should really be fetched from the module itself, and possibly just cached on the tx to be able to display selected protocol infos while the module is offline (or from within companion).

That should allow as well scenarios where only some protocols are enabled in the module, letting the TX know in advance so the user can only select from those.