pascallanger/DIY-Multiprotocol-TX-Module #196: Problems with s-fhss

michalelektryk, 2018-10-06:
I bought cheap Chinese airplane xk a700. It comes with its own controller, but not the great quality one.
So I thought I could use multiprotocol tx plugged into my turingy 9x to control it.
On the transmitter it says that it's s-fhss, so I rotated switch into 6's position, but I could not bind to aircraft, however I certain that my module works because I'm capable of steering my original tgy recovers without problem (they use flysky)

pascallanger, 2018-10-07:
The XK models perfectly work with Multi SFHSS. I have a XK600 and XK110.
Are you sure that you are really selecting the correct protocol?
Have you followed the right binding procedure on the plane (usually just press the bind button on the main board)?
Have you done the frequency tuning?

pascallanger, 2018-10-07:
"Plane going to autobind after powerup": this is most likely not the case with sfhss due to how the protocol is made... I would like to see close pictures from the main board.
Freq tuning: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md

pascallanger, 2018-10-07:
Do you have this board: https://www.motionrc.com/products/xk-a700-receiver
If yes then it's not using a SFHSS protocol since what I see on the picture is an A7105. SFHSS is based on the CC2500. Both have a different pinout and even if I don't see the ref on the RF chip it matches the A7105 pinout and not the CC2500.

michalelektryk, 2018-10-07:
Yes, this one.
I took apart transmitter and it uses a7105, but on the box it says s-fhss so I thought I is emulated somehow

pascallanger, 2018-10-07:
Nope...
This board looks so similar to the F9x9 wltoys boards which are using flysky.
Have you tried to bind it using flysky?
Power up the board then power on multi in bind mode (dial should be set to flysky before booting...).

pascallanger, 2018-10-07:
All I can say is that it's not the FUTABA SFHSS protocol. Someone needs to dump the TX SPI bus to reverse the protocol.

closed 2018-10-07 by pascallanger

michalelektryk, 2018-10-07:
Thanks for your help. Just last question. Could you give me any hints what's the fastest way to reverse enginner the protocol?

pascallanger, 2018-10-08:
You can check this link which shows you how to dump the SPI bus: https://www.deviationtx.com/forum/protocol-development/4595-spi-capture-guide
Once you have some dumps I can look at it.

michalelektryk, 2018-10-09:
For every one else with this problem. It's obviously an flysky, but after pairing use have to send special sequence on ch1-4, not sure what's exactly the sequence looks like, but for me putting all stricks in 1 extreme, a after that into the other worked