pascallanger/DIY-Multiprotocol-TX-Module #197: Eachine E010

Riffer, 2018-10-25:
It seems Eachine has changed the protocol again. Using MJXQ E010 on my X9D does not bind with my brand new E010 - the original remote works, I tested this.

Any chance to get this working again?

OpenUAS, 2019-01-08:
@Riffer Have you tested with self compiled around 20181025 (or later)? if so adding a Git SHA would help debug in for devs much easier.

Riffer, 2019-01-08:
@OpenUAS I must excuse for not mentioning. It seems to be a hardware issue with my module. Bought another one from BG and with the protocol works.

So, thanks for your question and case closed.

closed 2019-01-08 by Riffer

JurikRistjouw, 2022-03-10:
I had the reversed issue. I had an old  Eachine E010 and updated the firmware for my multimodule. sufficed to say, it didn't work like the wiki suggested to use H8_3D (https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#h8_3d---36). I Had to choose the "old" MJXQ setting with E10 subsetting and it worked instantly. Maybe you could choose H8_3D for the newer version. Good luck!