pascallanger/DIY-Multiprotocol-TX-Module #204: compilation error following tutorial

manotroll, 2018-11-23:
In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_678976\sketch\Multiprotocol.ino.cpp:1:0:

C:\Users\manot\AppData\Local\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.9\cores\maple/Arduino.h:32:6: error: previous declaration of 'void setup()' with 'C++' linkage

 void setup();

      ^

Multiprotocol:221:12: error: conflicts with new declaration with 'C' linkage

 void setup()

            ^

In file included from C:\Users\manot\AppData\Local\Temp\arduino_build_678976\sketch\Multiprotocol.ino.cpp:1:0:

C:\Users\manot\AppData\Local\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.9\cores\maple/Arduino.h:33:6: error: previous declaration of 'void loop()' with 'C++' linkage

 void loop();

      ^

Multiprotocol:496:11: error: conflicts with new declaration with 'C' linkage

 void loop()

           ^

Usando a biblioteca SPI na versão 1.0 na pasta: C:\Users\manot\AppData\Local\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.9\libraries\SPI 
Usando biblioteca EEPROM na pasta: C:\Users\manot\AppData\Local\Arduino15\packages\multi4in1\hardware\STM32F1\1.0.9\libraries\EEPROM (legacy)
exit status 1
conflicts with new declaration with 'C' linkage

and how to add this file to the list of selectable protocols?
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commit/cc2af73a057850cd0c379d5d6dd5bdb495dd2fb5#diff-d3d19a93898ce1c0331064919789325c

pascallanger, 2018-11-24:
You are not using the recommended version of Arduino 1.8.5 in the tutorial: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#install-the-arduino-ide

manotroll, 2018-11-24:
sure I did

closed 2018-11-24 by pascallanger