pascallanger/DIY-Multiprotocol-TX-Module #207: Receiver endpoints for DSMX on xlite do not go to 1000 - 2000

Kevc1, 2018-12-12:
When binding to the Mobula7 with DSM (both 2 or X) on the Xlite with the Vantac MPM module, the endpoints shown on the radio (988-1202) do not match what is shown in betaflight.  I am unable to get the endpoints for any channel to be 1000-2000. When i try to adjust the travel in the radio they go in the opposite direction of what i need.  Given this problem I am unable to use DSM.
If needed I can try it with other quads i have, let me know if you need that or any other info to resolve this problem.  I would really like to be able to fly all my quads using the xlite and I have lots of DSM quads. I am a Spektrum user trying to convert to OPEN TX.
Thx

Kevc1, 2018-12-12:
> Which version of MPM are you using? and which config for DSM?
> It's normal that the endpoints in microsec of the radio does not correspond to what you have at the receiver output since the MPM is defined to work with any radio. Instead -100/+100% on any radio will correspond to -100/100% of that specific protocol. In the case of DSM there are 2 choices depending in you've configured or not DSM_MAX_THROW as explained in the doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm---6 , check the 3rd bullet of the notes.
> I don't understand what you mean by: "When i try to adjust the travel in the radio they go in the opposite direction of what i need."
> I think you are confused... DSM does work correctly, you just need to read the doc and configure everything correctly.
> Pascal

Thx, I did see the entry on dsm_max_throw after submitting this. I’m not at home but I have the version as of this past weekend. Something like 1.1.2....22.
What I mean by adjusting the travel is: on the xlite when I try to change the endpoints I can only decrease the 100 and increase the -100, when I do that the numbers in beta flight for each channel go in the wrong direction. If I still have trouble I can be more specific with screen names when I have my transmitter in hand. 
Btw, I was able to get flysky working perfectly but I had to upgrade to the latest version.  Thanks much for the quick reply.

Kevc1, 2018-12-12:
One other question. I see option=number of channels from 4-12. Mine is at 6 which implies something is configured wrong. What would be an invalid option? What does options refer to ? Dsmx, dsm2, etc. just trying to understand what incorrect option I could have selected, thx again

pascallanger, 2018-12-12:
Ok you are confused as I said...
"What I mean by adjusting the travel is: on the xlite when I try to change the endpoints I can only decrease the 100 and increase the -100"
Obviously if you do that and like for any other models you are decraesing the range/throw which is what limit is used for you limit the range... 
As written in the doc I pointed you earlier and since you have left the DSM_MAX_THROW commented by default, to get 1000...2000 you must enable extended limits to get to +-125% on the OpenTX model.
Pascal

closed 2018-12-12 by pascallanger

Kevc1, 2018-12-12:
Will do, thx for the quick replies.