pascallanger/DIY-Multiprotocol-TX-Module #210: ia8x receiver unable to bind

sjb0095, 2019-01-05:
Hi, 

hoping someone can help!
v1.2.0.22 unable to bind to my receiver - (Taranis, IrangeX IR4X plus)
the receiver binds to the nirvana but I can not get it to bind to this module.

pascallanger, 2019-01-07:
Hi,

It would be better to ask the question on the main forum so everyone can contribute with their experience: https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page885

What's the LED status of the module when binding? (https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Troubleshooting.md#troubleshooting)
Have you tried to bind the RX at a distance? 
Have you tried different subprotocols since some receivers only support some modes? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#flysky-afhds2a---28

Pascal

sjb0095, 2019-01-07:
@pascallanger  - Thanks I've posted on the forum

LED Status: Blinking RED
Sub protocols - I've tried them all but PWM-Ibus works on another transmitter so it should bind to that.
Binding from a distance doesn't make any difference.

I've just noticed when pressing bind the Taranis is continues and you can't stop it from trying to bind.
FS A2 PWM,IBUS

pascallanger, 2019-01-29:
> I've just noticed when pressing bind the Taranis is continues and you can't stop it from trying to bind.
> FS A2 PWM,IBUS
That's normal since the multi module is waiting to receive an answer from the RX.
I don't have this specific receiver to test but I know that the FS-iA6B is working fine including telemetry.
Pascal

sjb0095, 2019-04-07:
Hi, just for info: I've tested a few other RX's and they work fine, there is something about this one that the module doesn't like.

DeADLoCK1911, 2019-04-28:
Actually the ia8x does not give out any telemetry as if you even bind it to a Fs i6 the binding screen won't go until u restart the transmitter so it basically just received and doesn't transmitts

DeADLoCK1911, 2019-04-28:
So now maybe you can close this issue

sjb0095, 2019-04-28:
> ia8x does not give out any telemetry as if yo

its not telemetry thats the problem - its can not bind to this module at all.

pascallanger, 2019-04-28:
@DeADLoCK1911 Are you able to bind this RX model?
The thing is that you must receive the RX ID to be able to pair with a RX.

DeADLoCK1911, 2019-04-29:
> @DeADLoCK1911 Are you able to bind this RX model?
> The thing is that you must receive the RX ID to be able to pair with a RX.

You just can't bind it as it does dot give out any information so the multi module just can't store it's rf id but you can bind it to fly sky radios

DeADLoCK1911, 2019-04-29:
@sjb0095 it's not only the telemetry but it doesn't give out any rf information out for the multi module to store . Even on the flysky radios the radio doesn't display successful bind like when you binded a fsia6 the binding screen disappeared after a successful bind.

pascallanger, 2019-05-01:
We could try 2 things:
@DeADLoCK1911: listen what the original Flysky TX is sending and look at the RX ID it sends
or simply try a dummy RX ID (0000000) and see if the RX reacts.
@sjb0095 : Can you try to bind a working RX and then without changing the RX_Num bind the ia8x? When you think the RX is bound, powercycle the Radio to exit the bind. Will the ia8x work after that?

DeADLoCK1911, 2019-05-04:
@pascallanger maybe a dummy Rx id is not a bad idea and might work btw im messing around with my flysky to see how that binds to the ia8x

AndreBL, 2019-05-04:
I'm using the multi module with an RTF AFHDS2A micro quad (QX65). The integrated RX does not send telemetry so the module gets stuck into bind mode until power cycled. After turning the power back on, bind is successful and everything works as it should, except telemetry of course because the RX doesn't have that feature.

AndreBL, 2019-05-04:
So I guess the RX ID may not be even checked by such telemetry-less receivers.

pascallanger, 2019-05-06:
@AndreBL Hummm so you are able to bind it with the multi module. Did you had another RX bound to the specific RX_Num before binding the ia8x?
I'm asking since if the bind does not complete a previously emply eeprom location would contain FFs and therefore the ID will be 0xFFFFFFFF. But if there was already a RX bound, the module would reuse this ID.
May be it doesn't work for @sjb0095 and @DeADLoCK1911 because 0xFFFFFFFF is used instead of a arbitrary value. If this is the case this is quite simple to solve but I need someone to run the tests.
Pascal

AndreBL, 2019-05-10:
Tested binding the QX65 quad with never used RxNums. Works fine. Just have to reboot the transmitter to get it out of bind mode.

AndreBL, 2019-05-10:
Just clarifying, it's not an ia8x receiver I'm testing... It's a micro quad which has an AFHDS2A RX integrated in it's flight controller board without telemetry capabilities.

pascallanger, 2019-05-10:
Multi v1.2.1.45
Modified AFDS2A to exit from bind if stopped from GUI.
IE: you don't need to power cycle the TX to exit bind anymore, just exit from bind. It's doing exactly like it has received the bind answer from the RX sending even the packets confirmation to the RX.
Could you test?

pascallanger, 2019-05-11:
@sjb0095 @DeADLoCK1911 Have you tested with this new version?

sjb0095, 2019-05-15:
> @sjb0095 @DeADLoCK1911 Have you tested with this new version?

Hey, Apologies I'm in the middle of moving house - I will test as soon as I can.

closed 2019-06-04 by pascallanger

DeADLoCK1911, 2019-06-10:
@pascallanger sorry for the late reply I have ordered a new ia8x will test as soon as I receive it

sjb0095, 2019-08-10:
@pascallanger @DeADLoCK1911  Hi Both, Sorry for the late reply - I a have now updated my module to the latest version and can bind to this RX. although I do have to bind it each time I power cycle it but I think that's the way this RX works. I actually don't like this RX and won't be using it in future but I'm sure this has helped others.