pascallanger/DIY-Multiprotocol-TX-Module #211: Selecting E01X Protocol causes MPM to stop working

MRC3742, 2019-01-08:
When the Protocol E01X is selected (or Custom # 45) my TX Radio Module Status display changes from v1.2.1.23 to No MULTI_TELEMETRY and the Red Module status LED goes out.
After this occurs, changing to any other protocol has no effect and the Module will not respond until a different protocol is selected and the radio is shut off and restarted.

I believe this is not an OpenTX issue as it occurs either using the protocol name or the custom options.

Using Taranis X9D+ with OpenTX 2.2.3
iRangeX IRX4 Plus with Firmware v1.2.1.23

Thanks - Rich

OpenUAS, 2019-01-08:
@MRC3742 Try with OpenTX 2.2.2 and nightly if possible just to rule it out...

MRC3742, 2019-01-09:
Using OpenTX 2.2.2 or current nightly firmware has the same results. 

Commenting out  "#define E01X_NRF24L01_INO"  in _Config.h file and updating MPM firmware causes TX Radio Module Status display to now change from v1.2.1.23 to "Protocol invalid" as expected and no longer needs to be rebooted to operate. Just changing the protocol will bring back the v1.2.1.23 and the MPM Status LED operates as intended.

@pascallanger  The E01X Protocol seems be causing the problem?

pascallanger, 2019-01-09:
I'll look into it soon.

pascallanger, 2019-01-09:
Ok so the issue is appearing when option=0. So either update to the latest version or set option to anything different of 0...
Check the protocol statement, this protocol is not working properly at the RF emulation layer and this is why option was introduced as an attempt to fix it: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#e01x---45 .  I don't have access to a receiver so nothing I can really do remotely.
Pascal

MRC3742, 2019-01-09:
Thanks Pascal

Using version 1.2.1.25 the MPM no longer stops working when either E01X or Custom 45 is selected.

I also do not have any way to test these new protocols but my issue was if searching for an unknown model protocol, by trial and error method, once I was to select the E01X Protocol the MPM would stop working entirely. So after that you would think you were checking other protocols but the MPM was no longer responding. 

Your update fixes this problem, although now you need a tester to help get the rest sorted out.

Thanks Again - Rich

closed 2019-01-11 by pascallanger