pascallanger/DIY-Multiprotocol-TX-Module #213: DSMX Channel Limitations

szempruw, 2019-02-18:
Sorry if this is the wrong place to ask. But why are the number of DSM2/DSMX servo channels limited to 6? I use module on Taranis X9D+. OpenTx 2.2.1 There are Spektrum Rx's  with more, i.e. AR930 has 9 channels and uses DSMX Spektrum transmitter. Is this only driven by OpenTx??? Or is synergy between OpenTx and RF module needed?

pascallanger, 2019-02-18:
The number of channels is selectable by software from 6 (default) to 12.
Here are the protocol details: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm---6
Some receivers don't like some number of channels , this is why you can force it. But if you bind using autodetect, the module sets the radio settings based on what the RX requires.
In short, you have to set the number of channels.
Pascal

closed 2019-02-18 by pascallanger

Jaubertel, 2019-02-18:
Hello

6 channels is only with PPM signal open tx   → external module.  With serial port is more. 

Provenance : Courrier pour Windows 10

De : szempruw
Envoyé le :lundi 18 février 2019 20:03
À : pascallanger/DIY-Multiprotocol-TX-Module
Cc : Subscribed
Objet :[pascallanger/DIY-Multiprotocol-TX-Module] DSMX Channel Limitations(#213)

Sorry if this is the wrong place to ask. But why are the number of DSM2/DSMX servo channels limited to 6? I use module on Taranis X9D+. OpenTx 2.2.1 There are Spektrum Rx's with more, i.e. AR930 has 9 channels and uses DSMX Spektrum transmitter. Is this only driven by OpenTx??? Or is synergy between OpenTx and RF module needed?

pascallanger, 2019-02-18:
This is wrong... PPM and serial can do exactly the same. You need to configure the number of channels in both modes as you want.