pascallanger/DIY-Multiprotocol-TX-Module #214: JJRC H31

500mania, 2019-02-28:
Witch protocol may be used for jjrc h311
[image]

pascallanger, 2019-02-28:
Is it H311(text) or H31 (title)?
If H31, it's not supported since we never had a good SPI dump: https://www.deviationtx.com/forum/protocol-development/6663-protocol-for-jjrc-h31?start=40
Pascal

pascallanger, 2019-02-28:
People are changing the board on this quad: https://youtu.be/-OuLu8bkGkQ
https://www.amazon.com/gp/product/B07BBN1S62/ref=as_li_qf_asin_il_tl?ie=UTF8&tag=jasonboyle-20&creative=9325&linkCode=as2&creativeASIN=B07BBN1S62&linkId=1d634bfb6f5642bbd7df19780463d244
Pascal

closed 2019-02-28 by pascallanger