pascallanger/DIY-Multiprotocol-TX-Module #216: CORONA v2 protocol / redled flashing / invalid protocol ?

Lolo-che, 2019-03-27:
Hi All, I need some help please !

My set is : 9RXPRO radio + 4in1 jumper multi module / serial mode. Software is updated.

I want to work with a corona V2 DSSS receiver, so I select the appropriate protocol in the radio menu, but then I get the red led on the 4in1 module flashing (= "invalid protocol selected" as described there : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Troubleshooting.md).

What can be wrong ?

All works perfectly with a DSM2 protocol for example...

Than you !

pascallanger, 2019-03-28:
What about upgrading the multi module?
Which version is it running right now?
Pascal

Lolo-che, 2019-03-28:
Hello,

OK, after upgrading the multi module the led is fix, so it may work (can't test right now - I'll let you know when done).

So much information to get, I didn't think to upgrade the module !!! For those who want to upgrade, just follow this : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md

Thank you again !

closed 2019-03-28 by pascallanger

Lolo-che, 2019-03-28:
Now works perfectly with my corona receivers !
Thanks