pascallanger/DIY-Multiprotocol-TX-Module #219: WFLY WRF07S does not bind

gian1312, 2019-04-21:
Hi,
If I choose Mult WFLY Channel 1-16 on a Jumper TX16 with the JP4in1 module it does not bind. Am I doing something wrong or is there another way to bind it? Or did I mess something up and it is not supported at all?

Best
Gian

pascallanger, 2019-04-26:
WFly on a WFR07S is working fine and has been tested by multiple users.
Is your module running the latest version?
What is the LED of the module doing? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Troubleshooting.md

gian1312, 2019-04-27:
@pascallanger thanks for your answer. Green is always on red is flashing fast when I press bind and stops as soon as I turn on the bind mode on the receiver. After that, both are solid but the reciver indicates to have no connection. Same outcome if I set the reciver to bind mode first. As module status, v1.2.1.25 is displayed. This seems to be the latest version?

pascallanger, 2019-04-27:
Latest version is 1.2.1.38 please upgrade first.
I'm closing this thread since you are double posting on rcgroups.

gian1312, 2019-04-27:
Thanks. Since version 1.2.0 is marked as the latest release I have to build from source and assume that the master branch is a stable version?

pascallanger, 2019-04-27:
yes

gian1312, 2019-04-27:
Thanks, I have now the firmware updated. It still does not work.
Following [Picture](https://www.dropbox.com/s/vgrf6tadefb1pre/20190427_163255.jpg?dl=0) shows my settings. Is there anything completely wrong with it?

pascallanger, 2019-04-28:
I'll try my WFly receivers when I'm back from vacation in a week.

gian1312, 2019-04-29:
Thanks a lot! Enjoy your vacation.

pascallanger, 2019-05-06:
I confirm that WFLY works correctly on Multi v1.2.1.32.
Here is a video showing how to bind a WFLY WFR07S and WFR06S: [url]https://youtu.be/Oj5Mx84l3Io[/url]
The procedure to bind:
 - Set on the radio Multi, WFLY (module LED should be steady ON)
 - power on the receiver (RX LED is red=no signal) and then press the set button until the RX LED flashes orange (=wait bind)
 - launch a bind on the radio (module LED should be blinking), wait for the bind to complete or stop it (module LED should be steady ON)
 - the RX should now respond to commands (RX LED is off=OK)

If it doesn't work for you, please report like I've just did on what the different LEDs are doing.

gian1312, 2019-05-07:
Thanks a lot! I won't be home the next two weeks. But I will let you know as soon as I am home and tried it. Just to make sure. If you talk about module LED, we are speaking about the red one? The green one is just for power?

pascallanger, 2019-05-07:
Module LEDs functionnality depends on which module model you have. But anyway I'm talking about the status LED the one which blinks.

gian1312, 2019-05-18:
So, I am finally back home again. 
I am using the stock 4in1 Module from a Jumper T16.
1. The radio is set to: Mult Wfly (V1.2.1.38)
    Ch1-Ch16
    Reciver Nr9
    Red and Green LED are steady on.
2. I power on the receiver. Red status led.
    Press set button until it flashes orange.
3. Start bind on remote
    Module:
    Red led starts flashing for a second. Green LED stays steady
    Receiver:
    Stops flashing orange and gets black for 1-2 seconds. After that, the receiver led is steady red.
4. Module LED: Red and green steady

Same behaviour on three WFLY WFR07s receiver
I hope, this are the information you need. If there is anything else I could do, just let me know.

pascallanger, 2019-05-18:
That's strange... Can you try:
- Download the latest firmware
- Edit the file WFLY_cyrf6936.ino and uncomment (remove the //) on this line which is at the begining: //#define WFLY_FORCE_ID
- Flash the firmware
- Try binding again

gian1312, 2019-05-18:
Thanks! I will try to do that.
Sadly,
`		rc_ch_num=0x15;						// RF channel to send the current hopping table` 		
Is not defined in this scope. (File WFLY_cyrf6936.ino), line 256. 
What datatype would it be and where is it defined?

pascallanger, 2019-05-18:
typo.. it should be rf_ch_num

gian1312, 2019-05-18:
Thanks!
But it is still the same behaviour. Everything seems to be fine, but after approx. two seconds, the receiver turns red again.

pascallanger, 2019-05-18:
Ok so it must be something with the Jumper 4in1 module inaccuracy again. I'll try on my side since I've just received my T16 but the weird thing is that it binds.

gian1312, 2019-05-18:
It could be a faulty module as well. I just have one though, so I can't check. Since there are several different issues with the T16 I have to send it back anyways. I will have them replacing the module as well. It will take quite a while though. Just to make sure. You haven't changed anything in the config file beside uncommenting multi_telemetry and commenting multi_status? (Needed to be ably to use telemetry on the SxR receivers of FRSky)

pascallanger, 2019-05-18:
Yes no other changes.

pascallanger, 2019-05-18:
What's wrong with your T16?

gian1312, 2019-05-18:
Broken roller (I have fixed it for now, but as soon as the snow is gone and my flying fields are closed (Cows), I will send it to my reseller to get it fixed properly.) and interferences of some switches.  SE triggers SF and vice versa. But just with certain combinations.
Otherwise, it is a nice remote though. 
And am I right if I assume, that the SxR configuration Lua scripts are not working with Multi Module and JumperTX? [According to issue 136](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/136)

pascallanger, 2019-05-18:
Correct

pascallanger, 2019-05-18:
Ok, can you try the latest version 1.2.1.51?
In the _config.h, you need to enable the #define USE_CYRF6936_CH15_TUNING .
Compile, flash, bind and test to see if the RX likes the freq tuning by assigning a pot to channel 15.

pascallanger, 2019-05-18:
I've just tested and I confirm that if I enable the setting, set CH15=100% S1 and then turn S1 I'm loosing connectivity to the RX (LED red) on both extremes (not full extreme but nearly). I can only see that my JP4in1 module is not bad since the frequency looks centered.

pascallanger, 2019-05-24:
@gian1312 any news

gian1312, 2019-05-25:
@pascallanger I am really sorry, I haven't seen the e-mails. Thanks a lot! I will try it today.

gian1312, 2019-05-25:
It is similar as before (a lot of extremly short binds.), but at 40% 62% and 80 %, 100% I had binds which lastet up to 10 seconds and I could even move servos. I am wondering, if it would be possible to go higher than this +300. Where in the code would I find this value?

pascallanger, 2019-05-25:
Have you tried to enable extended limits and set the output to 125%?

gian1312, 2019-05-25:
I did this now. Interestingly, I got a few seconds bind with servo movement at -145% and at approx +120% aswell.

pascallanger, 2019-05-26:
Can you try to change in the file WFLY_cyrf6936.ino (line 141):
`		if(Channel15!=Channel_data[CH15])
		{ // adjust frequency
			Channel15=Channel_data[CH15]+0x155;	// default value is 0x555 = 0x400 + 0x155
			CYRF_WriteRegister(CYRF_1B_TX_OFFSET_LSB, Channel15&0xFF);
			CYRF_WriteRegister(CYRF_1C_TX_OFFSET_MSB, Channel15>>8);
			Channel15-=0x155;
		}
`
by
`		if(Channel15!=Channel_data[CH15])
		{ // adjust frequency
			**Channel15=(Channel_data[CH15]>>2)+0x155;**	// default value is 0x555 = 0x400 + 0x155
			CYRF_WriteRegister(CYRF_1B_TX_OFFSET_LSB, Channel15&0xFF);
			CYRF_WriteRegister(CYRF_1C_TX_OFFSET_MSB, Channel15>>8);
			**Channel15=Channel_data[CH15];**
		}
`
Please try to turn the pot slowly but also rebind where it seems to work for a few sec.

gian1312, 2019-05-26:
Thanks a lot! I am really sorry, but I am away until next weekend.
I will let you know as soon as I am back home again.

gian1312, 2019-06-01:
I have tried it. I made steps of +-10% and rebound it every time. Above 50%, some short binds were possible (with servo movement).
There were several short binds without movement over the whole spectrum.
There are two other strange things. The module LED stops flashing really fast. The remote contiues with the binding sound. 
The RC signals bind, where as the receiver continiues with blinking. Or the receiver stops blinking but the remote continues with the binding sound. 
It is probably a faulty module.

pascallanger, 2019-06-03:
I'm not sure what to do next... I can't reproduce your issue on the T16 with the JP4in1.
Are you able to send me one of your RX so I can test?

gian1312, 2019-06-03:
I could do this. (However, I have to check the shipping costs first. They have to be more or less reasonable. Where are you living?)
But since you already confirmed, that you can bind a WFR07S with a Jumper 4in1, we know that the software works.
Further, we know that my 4in1 can't bind to three different WFR07S. My WFly remote can bind the same three WFR07S without problems. Thus, the RX work as well. 
Thus, I think I got a faulty 4in1 module. 
I will send my remote back to the shop soonish.
However, if you think it would help you to improve the multiprotocol project, I can send you the receiver.

pascallanger, 2019-06-03:
Have you tried when it's working for a couple of seconds to power cycle the TX? I'm wondering if the RX will start to work again for a couple of seconds.

gian1312, 2019-06-03:
No, it doesn't. If you manage to power it off fast enough and power it on again it doesn't bind again.

closed 2019-06-04 by pascallanger

gian1312, 2019-09-29:
Just a follow up. I finally managed to get my 4 in 1 module replaced. With the new module, it works perfectly fine out of the box. 
Thanks a lot for your work. It is an awesome work you are doing here with the multiprotocol firmware.

pascallanger, 2019-09-29:
Great!