pascallanger/DIY-Multiprotocol-TX-Module #221: WFly FR209S dont bind

backman8571, 2019-04-29:
I have the 2.2.3 Opentx release and the new T16 radion with Jumper 4in1

I will not bind my Wfly RF209S receiver, no issue with my Wfly T12 radio

What can be done to fix this issue

Per

pascallanger, 2019-05-06:
Double post with rcgroups. This is not the original WFly protocol but a Futaba protocol using an A7106.

backman8571, 2019-05-06:
So what interface should i use then ??
Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

pascallanger, 2019-05-06:
See RCGroups, this is where you are asking tons of questions!!!

backman8571, 2019-05-06:
So why can you not be nice and help, since i dont know, is that wrong
asking for help
Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

pascallanger, 2019-05-06:
I'm nice and helping you rcgroups. I simply don't see the point on answering here and on the forum...
You wrote on rcgroups:
> It works fine with my Wfly T12 radio no problem and the Flysky receiver's dont bind with the T12, T12 is a copy of Futaba MZ18 or MS18 a great radio but love to get all interface's to work in one shell.This is a new protocol and surely not the real wfly protocol.2 solutions:- you dump with a logic analyzer the SPI bus on the transmitter and send the files- ship your transmitter and receiver for analysis to Goebish or myselfPascal | And it works fine with my Wfly T12 radio no problem and the Flysky receiver's dont bind with the T12, T12 is a copy of Futaba MZ18 or MS18 a great radio but love to get all interface's to work in one shell.
And I've replied:
This is a new protocol and surely not the real WFLY protocol.
2 solutions if you want this protocol to be reversed:
- you dump with a logic analyzer solder to the transmitter the SPI bus and send the files for analysis
- you ship your transmitter and receiver for analysis to Goebish or myself and we will return them once the reverse engineering is done
Pascal

backman8571, 2019-05-06:
Pascal

Thank you for the quick reply

Well i live in Dallas texas so it be high cost to ship to you

I dont like to solder stuff on a new unit particular since i am not sure
what i am doing and have the needed equipment to get the dump for you.

best right now is trying to find someone local that can help me flash the
4in1, that news how to compile.

After that i can try it again and see.

Is it a way to do a backup of the existing firmware before i would blew the
old one away ??

Alsowill it be a tool that can connect to the usb port on the 4in1 module,
if it is today please send me a link i have not seen that and i do
apologize of not be to technicaly engineering.
Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

pascallanger, 2019-05-06:
Upgrading the module will not help in the case of this wfly/futaba protocol since it's a new protocol that we haven't seen yet.
For the upgrade, once again I've replied on rcgroups to your exact same question? Where should I answer? I don't want to be inpolite but you are using my valued time to reply twice to every of your questions while I could work on other protocols...
So again, you don't need anything special to update the firmware on your Jumper 4in1 module. You just need a simple usb cable like you used to use on your mobile phone and if I'm not mistaken even included in the box. There is no way to break the module. And once again read the documentation, everything is explained.

backman8571, 2019-05-06:
Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

backman8571, 2019-05-06:
Pascal No you are not impolite.

When i check on the wiki the only thing i see is a procedure where you
solder 4 pins to back of the module and the using an std32 arduno upgrade
tool

So if all i need is a mini usb to usb cable and windows software then that
is great news.

I probably still need to contact someone here for how to compile the 4
module's into one, i did not understand that process, sorry its probably
written well but i am no smart in compiling software, sorry for that

I rather communicate here with you then on the RC-Group's, after i have it
figured out i will do an update and referencing your help

I just need to take baby step's

The process of flashing the radio is simple you start the open tx tool
download the firmware and point to the firmware in the tool and click write

Is the process similar after you have compiled the firmware, or as part of
compiling will that upload to the module, i am confused in that matter

1. is it the same tool (arduni) to be used when you use the usb interface
on the module
2. need some help feder on the process of putting the data together and
flash all 4 at the same time (confusing to me sorry)
3. burn to the 4in1 unit i hope its as simple as for the opentx tool

i also used the frsky update sport tool for flashing receiver's

Thats all my knowledge in flashing unit's not great, again sorry for being
a pain

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

brandond, 2019-05-06:
As he said above, nobody's dumped the protocol yet, so regardless of whether or not you can figure out how to accomplish the relatively simple process of reflashing your TX module with the provided USB cable, there will not be any working firmware to put on it.

Someone's got to open up the controller, solder on some test leads, and dump out the commands going into the RF chip (A7106). Once that's done someone can add support for the new model, and you can try flashing it onto your TX.

I'm not sure where goebish is, but I'm located in Oregon and do have a logic analyzer, if you want to trust sending your hardware to a stranger on the internet who's not overseas.

Honestly if you don't want to use the T12, why not just buy another RX that already works with your 4-in-1 TX? It's not like they're expensive.

backman8571, 2019-05-06:
I did not say I dont like it , my intention was to have one go to the
Jumper TS16 and for some of my electric small fixed wings I like the T12 by
wfly the gimbals feel better

I will reconsider shipping my wfly to you will probably be this coming
weekend

To busy before that

Per

pascallanger, 2019-05-07:
To flash the module it's simple and really just read the doc will get you going.
If you read the first few lines you will see this note:

> The latest jumper modules have an integrated FTDI appearing as a CP2102 device on the computer. You therefore don't need the FTDI adapter below and don't need to open your module to flash it.

As indicated, you don't need to open or do anything apart than connect the module to your computer using the USB cable.
The steps to flash are the following but you still need to read the documentation for each step:
 - Install and configure the Arduino IDE
 - Download the source code
 - Configure the module features to your liking by editing the _config.h file
 - Load the code in Arduino IDE, select the right interface and com port
 - flash and done
This is really simple but you need to read the documentation which will guide you through the steps.

backman8571, 2019-05-07:
Question

The new FlySky version FS 2A.

I dont see that in all protocol's on github, im looking at version 1.2

Am i doing something wrong ??

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

pascallanger, 2019-05-07:
I don't understand the question... It's documented here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#FLYSKY-AFHDS2A---28
And is in the code: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/AFHDS2A_a7105.ino
What are you trying to do??

backman8571, 2019-05-07:
Oh, i was looking at my Radio the new Protocol only say FS 2A not the
complete AFHDS2A

Ok txs

Im trying to follow the instruction's and was just checking that all
protocol's im interested in was there

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

backman8571, 2019-05-07:
So this is the error im getting

Sketch uses 84964 bytes (70%) of program storage space. Maximum is 120832
bytes.
Global variables use 4192 bytes (20%) of dynamic memory, leaving 16288
bytes for local variables. Maximum is 20480 bytes.

Looking for Maple device ...
Maple USB Device found.

Attempting to flash module via Maple USB ...
java -jar maple_loader.jar COM6 2 "1EAF:0003"
"C:\Users\Per\AppData\Local\Temp\arduino_build_128593\Multiprotocol.ino.bin"
maple_loader v0.1
Resetting to bootloader via DTR pulse
Searching for DFU device [1EAF:0003]...
dfu-util - (C) 2007-2008 by OpenMoko Inc.
This program is Free Software and has ABSOLUTELY NO WARRANTY
Couldn't find the DFU device: [1EAF:0003]

Done.

Any idea

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

pascallanger, 2019-05-07:
> Oh, i was looking at my Radio the new Protocol only say FS 2A not the complete AFHDS2A
The radio GUI is independant of the module. Even if you pull out the module the menu will still give you the same list of protocols. Which means that the presence in the menu of a protocol does not mean that the protocol is in the module. The telemetry status where the version of the module is displayed is giving this information.

pascallanger, 2019-05-07:
> Looking for Maple device ... Maple USB Device found. Attempting to flash module via Maple USB ... java -jar maple_loader.jar COM6 2 "1EAF:0003"
The iRangeX module is different from the Jumper as stated in the doc. You must write the bootloader before being able to use the USB port (don't ask me why the manufacturer is not adding the bootloader at production time). To burn the bootloader you must solder the pins and use a FTDI to program the module once for all.
Check the doc again it's all explained.

backman8571, 2019-05-07:
Ah, had a felling that was why.

I will try that now

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

OpenUAS, 2019-07-31:
@backman8571 No new info added, I'll assume it worked for your.
 @pascallanger would you be so kind as to close the issue?

pascallanger, 2019-07-31:
The subject is still true. WFly FR209S does not bind since it's not a supported protocol.
I've received some hardware to play with but no time so far to look at it.
Closing as unsupported for now.

closed 2019-07-31 by pascallanger