pascallanger/DIY-Multiprotocol-TX-Module #224: Found correct protocol for DHD D4

marcfon, 2019-05-05:
I found that the [DHD D4](https://www.banggood.com/DHD-D4-Mini-Pocket-Drone-WIFI-FPV-With-720P-Camera-Altitude-Mode-RC-Drone-Quadcopter-p-1210848.html?ID=22342482&cur_warehouse=CN) binds with the Bayang STD protocol. 

There's just a weird issue with the throttle. When you throttle up or down there is a lag of a few seconds. This doesn't seem to be the case with the other input controls.

pascallanger, 2019-05-06:
Humm a new version of the bayang protocol? We should be able to figure out what's going on if you still have the original TX but this will take me a couple of days before coming back to you...

marcfon, 2019-05-07:
I do have the original TX. And there's no rush from my side.

Would be glad to help out in any way I can so just let me know what you need from me.

By the way it's actually a really fun quad to fly! :)

pascallanger, 2019-05-08:
With the new Multi version v1.2.1.40, we can now look at your DHD D4 payload and see what's going on.

How to do it:
- create a new model on your radio with a custom protocol number 63, sub_protocol 1 (for 1Mb), RX_num=0 (scrambled) and option=-1 (scan RF channels)
- download the latest code, edit the file multiprotocol.ino and uncomment the line (remove the //) at the begining of the file: #define DEBUG_SERIAL
- connect the USB cable to your module and upload as usual the firmware but don't disconnect the USB cable
- open in the arduino ide the serial monitor: Tools -> Serial Monitor
- you will see a new window with a lot of information displayed and the module will start scanning (option=-1)
- power on your original TX and quad, wait for the bind to finish
- now look at the new window, you should find some valid frames being displayed for some channels
- choose one of the channel and set the value of option with the value
- you'll now start to see on the serial monitor window a lot of packets being displayed on this specific channel only
- play with throttle for a couple of seconds
- send me a copy and paste of the content of the full window (ctrl+a to select everything and then ctrl+c to copy)

To use you module normally, you must edit the file multiprotocol.ino and comment the line at the begining of the file (add //): #define DEBUG_SERIAL .

Waiting for you to try it out, Pascal

marcfon, 2019-05-09:
I've managed to get the debug output! It's quite a lot so I uploaded them here.

Method 1 (one key take off / landing)
https://pastebin.com/mpV3WZrZ

Method 2 (throttle to arm (props start spinning / throttle up to take off)
https://pastebin.com/5K7FiaLP

Not sure if it's relevant but I seems that there were 4 channels active during the scan.
Channel=72,0x48
Channel=74,0x4A
Channel=75,0x4B
Channel=80,0x50

pascallanger, 2019-05-09:
The Bayang protocol is using 4 channels so yes the 4 that you've found are the ones being used by your original TX.

I can see in your dumps:
 - packet period seems to be around 7770µs on channel 74, can you confirm that this is the same on other channels (it should be the same but just checking)? => Bayang has a period of 8000µs to come back on the same channel, that could explain the lag if the RX is strict but I doubt but why not...
 - packet[1] = 0xF4 -> Multi is sending 0xFA but your original TX must have different rates (low/medium/high or easy/expert or...) Can you switch to the highest rate to see which value you are getting on this byte?
 - packet[3] switches between 0x00 and 0x20 when you press the one key take off/landing which is accessible through a switch on CH12. Have you tested it?
 - packet[12] and [13] have a fixed value of 0x37 and 0xED which is not what Multi is currently sending (0xxx and 0x0A)

Can you dump the bind phase at the original TX power up on channel 0? It would be good to get this info as well.

The next phase when you'll have completed the couple of actions would be to add a new subprotocol to implement the differences and test if it does improve.
Pascal

marcfon, 2019-05-09:
Okay here's some additional info!

Let me add to this again that the lag only occurs with the throttle. The other controls respond as expected. I can make a video to show what happens if that helps.

Channel scan results.
https://pastebin.com/Tp2UxUtu

Bind phase. (Channel 0  = RXNum 0??)
https://pastebin.com/TBp5G7nC

There is rate setting on this drone (as far as I know). There are buttons fo video recording, photo taking, an emergency stop, headless mode and flips.

pascallanger, 2019-05-09:
Channel 0 is option=0. Your dump of bind phase has been done on channel 74 so it has only normal packets... Can you redo it?

> There is rate setting on this drone (as far as I know).

Can you press the button (multiple times if needed) on the TX to make it go to highest rate and then dump a normal packet (option=74)?

marcfon, 2019-05-09:
> Channel 0 is option=0. Your dump of bind phase has been done on channel 74 so it has only normal packets... Can you redo it?

Got ya! Here it is.
https://pastebin.com/g95pkhcF

> Can you press the button (multiple times if needed) on the TX to make it go to highest rate and then dump a normal packet (option=74)?

I meant to write there isn't a rate option.
Here is a [photo of the TX](http://www.hobbywow.com/desc/HR/rc3358/HR3358_24.jpg) with all the options.

pascallanger, 2019-05-09:
What does the accelerate and decelerate buttons do? isn't it rates? I see on BG this in the description: "It has 3-level flight speed to switch that can make more fun with the flying." which looks like a description of rates...

pascallanger, 2019-05-09:
Please try v1.2.1.41 using a custom protocol=14, sub_protocol=4, option=0.
Let's see if it fixes your throttle issue...

marcfon, 2019-05-09:
> Please try v1.2.1.41 using a custom protocol=14, sub_protocol=4, option=0.
> Let's see if it fixes your throttle issue...

Yes it does! The strange thing though is the STD Bayang subprotocol now also seems to do what it's supposed to do. 

I've tested both 14/0 and 14/4 and there's no any noticeable difference. Was this a user error on my side?

You're totally right about the rates! I didn't know that. Do you still want me to have the dump for that?

And a question related to this... would it be possible to use a 3 position switch to simulate the +/- buttons on the original TX? I know that this is rather an OpenTX question but maybe you have some experience with this.

marcfon, 2019-05-09:
And to add to this I think it's good to share how that regular take off procedure works for this drone (so not the one key take off). 

You have to throttle all the way up to arm. Then move the throttle back to the **middle**, now the props will start spinning. From here you throttle back up again to take off.

pascallanger, 2019-05-09:
> Yes it does! The strange thing though is the STD Bayang subprotocol now also seems to do what it's supposed to do.
> I've tested both 14/0 and 14/4 and there's no any noticeable difference. Was this a user error on my side?

This doesn't make any sense. Everything I've implemented for your D4 is conditionned on choosing 14/4. 14/0 is basically the same as before...

> You're totally right about the rates! I didn't know that. Do you still want me to have the dump for that?
Yes like this we know the values.

> And a question related to this... would it be possible to use a 3 position switch to simulate the +/- buttons on the original TX? I know that this is rather an OpenTX question but maybe you have some experience with this.

We usually always set high rates since on the radio you can apply some expo and dual/rate if you think it's too fast.

marcfon, 2019-05-09:
> This doesn't make any sense. Everything I've implemented for your D4 is conditionned on choosing 14/4. 14/0 is basically the same as before...

> Yes like this we know the values.
>
>
> We usually always set high rates since on the radio you can apply some expo and dual/rate if you think it's too fast.

pascallanger, 2019-05-09:
???

marcfon, 2019-05-09:
> ???

Accidentally closed it. And now reopend again :-/

marcfon, 2019-05-10:
> Can you press the button (multiple times if needed) on the TX to make it go to highest rate and then dump a normal packet (option=74)?

Here is the dump file for the different rates. I switched from low to high during flight.
https://pastebin.com/ibjyJ1Sf

Is this the info you were looking for?

pascallanger, 2019-05-10:
Can you try Multi v1.2.1.46?
The quad should be in high rate.

marcfon, 2019-05-11:
> Low rate is 0xF4, Mid is 0xF7, High is 0xFA.
> Can you try Multi v1.2.1.46?
> The quad should be in high rate. If it works I won't touch it anymore.

I can confirm that the quad is in fast mode on v1.2.1.46.

Now I'll google how to set different rates using OpenTX :)

Thanks a lot for the help Pascal! Much appreciated.

closed 2019-05-11 by pascallanger