pascallanger/DIY-Multiprotocol-TX-Module #226: Missing Support/Documentation for "iRangeX IRX4 Lite"

boennhoff, 2019-05-23:
Aloha,
I bought an _iRangeX IRX4 Lite_ (from BG for the FrSky X-Lite Transmitter) and would love to flash it to the latest _Multiprotocol_ version, but I don't get it to work. It is working as of now, and I need it for my Whoops so I am very careful to not brick the thing.  :-)

My actual questions are:

### Flashing from TX?
Where do I find the Bootloader that needs to be installed on the X-Lite transmitter to eventually be able to do the famous _Flash from TX_? I found the *.app that needs to be run, but I can't load it with the default OpenTX Bootloader.

### Flashing over USB?
Connected over USB the board is recognized as a Maple Device (Dev Id similar to `1eaf:1003`) without flashing anything yet. So i assume there is a USB Bootloader (wrong?). But the method _Flash over USB_ is not working.
The Arduino IDE finds a Maple Device during _Upload_, utilizes the `dfu-util` that eventually fails (after a timeout)... The IDE strangely still acknowledges a successful flashing. The process was tested on Linux and Windows (USB pass-through to VM)).

### Flashing over Serial?
When trying to do the _Flash over Serial (FTDI) with Bootloader_, I also have no success. The LEDs flash differently than when trying the USB connection (only red LED blinking, instead of constant green and red blinking with USB). I also made sure that I really, really, really only use 3.3V when using this method. But the board actually asks vor 5V (see picture)... So should i try it out with 5V even if the Wiki is against that? Actually that is the only thing that I can think of, that I haven't tried yet...

What could I do? Any ideas? Is more information needed?
Sorry to bothering you guys and thank you for such an awesome product!

### Pictures of the HW
[image] [image] [image]

pascallanger, 2019-05-23:
> Flashing from TX?
> Where do I find the Bootloader that needs to be installed on the X-Lite transmitter to eventually be able to do the famous Flash from TX? I found the *.app that needs to be run, but I can't load it with the default OpenTX Bootloader.

Please check this part of the documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Flash_from_Tx.md#upgrade-the-bootloader-and-install-apps

> Flashing over USB?
> Connected over USB the board is recognized as a Maple Device (Dev Id similar to 1eaf:1003) without flashing anything yet. So i assume there is a USB Bootloader (wrong?). But the method Flash over USB is not working.

Can you elaborate on Dev ID similar? What's the exact Dev ID of your module?
It looks like the module has a bootloader but the firmware has not been compiled with USB serial...

> Flashing over Serial?
> When trying to do the Flash over Serial (FTDI) with Bootloader, I also have no success.

Unless Boot0 is grounded, you won't be able to update the firmware that way. When I look at your pictures, I don't see a trace going out from the boot0 pin but it could go under the stm32 so it's hard to say where boot0 is accessible (if accessible)...

Pascal

benlye, 2019-05-23:
Pascal beat me to it!

What he said, plus it would also be good to see the verbose output from the IDE when you attempt to upload.

boennhoff, 2019-05-23:
Thanks for the fast feedback. Somehow I have overseen that part of the [HOWTO](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Flash_from_Tx.md#upgrade-the-bootloader-and-install-apps), thanks for pointing me to it. The alternative Xlite Bootloader works (V3.0) now, and I can run the `FlashMulti_xlite.app`.

But if I try to flash `Multi-STM_TXFLASH_INV_OPENTX-v1.2.1.39.bin` (I shortened the name to be able), it returns with the following result:

    Flashing Complete
    FAILED
    0000 0000

The version of the module stays at 1.2.0.22.

A kernel log excerpt after connecting over USB:

    kernel: usb 1-3.4.3: New USB device found, idVendor=1eaf, idProduct=0004, bcdDevice= 2.00
    kernel: usb 1-3.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
    kernel: usb 1-3.4.3: Product: Maple
    kernel: usb 1-3.4.3: Manufacturer: LeafLabs
    kernel: cdc_acm 1-3.4.3:1.0: ttyACM0: USB ACM device

Of course I also tried the Flash over USB method again, here the log output of the Arduino IDE (i used a screenshot to make clear how much the IDE let it look like a successful upload)
[image]

The kernel log suggest that the Device ID is `1eaf:0004`, but in the Arduino IDE it refers to `1eaf:0003`... I checked that twice, I selected the right port. I am pretty confused now, well at least I made some progress and the thing did not yet release any magic smoke :-) Heads up!

Is this piece of HW new to you guys?

boennhoff, 2019-05-23:
Tested again in the IDE, and with the Transmitter turned off, the Serial Device was different and the result also:
```
Maple device found. Attempting USB upload.
/home/chris/.arduino15/packages/multi4in1/hardware/STM32F1/1.1.1/tools/linux/maple_upload ttyS0 2 1eaf:0003 /tmp/arduino_build_347318/Multiprotocol.ino.bin
dfu-util 0.8

dfu-util: Invalid DFU suffix signature
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Opening DFU capable USB device...
dfu-util: Cannot set alternate interface
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Waiting for /dev/ttyS0 serial...Done
```

Doesn't solve my confusion... Any idea on what's going on?

benlye, 2019-05-24:
Current thinking is that your module is running firmware without the USB code. To get it upgraded you may have to do some soldering so that you can connect an FTDI adapter, and somehow put the BOOT0 pin to ground (that will be the hard part).

I'll play around and see if I can reproduce your experience on a test board so we can be sure of what's going on. 

The manufacturers don't give us hardware to work on so we haven't seen this one before. 

Also, no one has asked about flashing this kind of module before, which makes me wonder if there's something strange going on with yours.

boennhoff, 2019-05-24:
> Current thinking is that your module is running firmware without the USB code. To get it upgraded you may have to do some soldering so that you can connect an FTDI adapter, and somehow put the BOOT0 pin to ground (that will be the hard part).

Where do you see the BOOT0 pin? I was looking for something like that, but couldn't determine it. I am open to tinkering and soldering, motivated but not a pro, so the complexity of the mod should not be overwhelming :smile: 

> I'll play around and see if I can reproduce your experience on a test board so we can be sure of what's going on.

Awesome, thanks! If you need more input just tell me.

> Also, no one has asked about flashing this kind of module before, which makes me wonder if there's something strange going on with yours.

That's interesting since I found comparisons between this one and the Vantac MPM module, which is more expensive but not shielded. I would have assumed it's a better hacking target.

pascallanger, 2019-05-24:
Ben, there are some talks around 1eaf:0004 and resolutions if you look on Google. But may be you already checked?

benlye, 2019-05-24:
The different USB IDs mean different things `1eaf:0003` means that the board is in DFU mode, `1eaf:1004` means that it is in USB / tty mode.  The flash routine is trying to get the device into DFU mode before flashing it.

benlye, 2019-05-24:
Essentially what happens is the script looks for any Maple device, `1eaf:0003` or `1eaf:0004`.  If either is present it runs the `maple_upload` script which first tries to reset the device to DFU mode by sending a magic sequence to the serial port (`upload_reset` is called).  I think this is where it's gong wrong and for some reason the board isn't resetting to DFU mode.

So, we expect to see `1eaf:1004` when the board is plugged in, and `1eaf:1003` momentarily when it's being flashed or if there is no firmware on it.

When I get a chance (may be tonight / tomorrow)  I'll write out a full set of command line steps that we can use to try to see exactly what's going on.

@boennhoff you could always try running `upload-reset` to see what happens.  Do `lsusb` before and after to see if you get different USB device IDs.  You can get the binary [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module-Boards/raw/master/source/stm32/tools/linux/upload-reset).  The syntax is like `./upload-reset /dev/ttyACM0 750`

boennhoff, 2019-05-24:
Okay, that enlightens the situation a little bit. I will definitely try that and report back.

I am also interested in the way the components play together, I can't get my head around it alone. How is the STM connected to the Radiochips? Where is the Bootloader located, are there different ones? Which component is taking over which part, booting, flashing, running applications. Why is there a Micro-USB port if the designer did not support any USB connection? What's the role of BOOT0?
Are there some explaining/teaching resources on that matter?

boennhoff, 2019-05-24:
Okay, the following happens (module is always in transmitter):

**Transmitter turned off, USB connected**
- Green LED steady, red LED blinking
- `lsusb` reports `1eaf:0003`
- `upload-reset` does not work because of missing /dev/tty*

**Transmitter turned on, USB connected**
- Green LED steady, red LED steady
- `lsusb` reports `1eaf:0004`
- `upload-reset /dev/ttyACM0 750` turns off green LED, red LED flashes, ~2s later the module seems to reboot, `lsusb` still reports `1eaf:0004`, a second try leads to an error because of missing serial device. reconnecting USB allows to repeat that step.

Hmmm, that result doesn't look like the expected one, does it?

benlye, 2019-05-24:
> How is the STM connected to the Radiochips?

Pascal needs to answer that one :)

> Where is the Bootloader located, are there different ones?

The bootloader is stored in the first section of flash memory in the STM32 chip.  We only use one, which is customized to support the USB connection and also flashing from the TX.

> Which component is taking over which part, booting, flashing, running applications. 

The bootloader does most of that - it boots then hands over to the application.  When the bootloader is running the chip is in DFU mode, which is the state it has to be in to flash it.  The LED blinks rapidly when the bootloader is running.

> Why is there a Micro-USB port if the designer did not support any USB connection?

To make the USB port work two things are required - the USB-enabled bootloader, and application code which includes support for USB.  That your board is reporting as a `1eaf:0004` when plugged in actually suggests that you have both.  If you did not have the USB code the board would stay as `1eaf:0003`.  Most of the Multi board manufacturers do not ship their modules with the bootloader installed.

> What's the role of BOOT0?

Connecting BOOT0 to ground puts the board in a state where firmware can be written to it over the serial connection.  If BOOT0 is not grounded it is simply not possible to flash over the serial port.  Think of it as a failsafe - if we completely trash the bootloader and firmware we can use BOOT0 mode to recover the device via an embedded flashing method.

> Are there some explaining/teaching resources on that matter?

Probably the [STM32 datasheet](https://www.st.com/resource/en/datasheet/stm32f103c8.pdf).

benlye, 2019-05-24:
OK, those results are interesting.  It seems that powering the module from the TX is doing something interesting to it, and when we try to reset it to DFU mode it just switches back to application mode.  Powering it only via USB seems like the way to go.

Can you try:
* Power the module from USB only
* Run dfu-util to list the DFU devices: `dfu-util -l`

You'll find dfu-util in `/home/chris/.arduino15/packages/multi4in1/hardware/STM32F1/1.1.1/tools/linux/`

If you get output like `dfu-util: Cannot open DFU device 1eaf:0003` try prefixing the command with `sudo`.

benlye, 2019-05-24:
I think I may know what the problem is - permissions on the device.  If the list command works with `sudo` and fails without, it's a simple fix.

benlye, 2019-05-24:
Assuming my theory about permissions is right, run this command:
`sudo cp /home/chris/.arduino15/packages/multi4in1/hardware/STM32F1/1.1.1/tools/linux/45-maple.rules /etc/udev/rules.d/`

Then unplug / plug the module (keeping the TX off).

That udev rules file tells the system to set more permissive permissions on the Maple device when it is connected, both in DFU and USB mode.  

On my system, where dfu-util wouldn't work without `sudo` the rules fixed the problem and I was able to flash using the `maple_upload` script.

boennhoff, 2019-05-24:
Doing the commands by hand is a good idea, I already stumbled about permissions in the first place, so maaaaaybe running them as sudo resolves my issue. But the `45-maple.rules` file won't help, it resides in the right place after my first attempts already -- the suggestion is somewhere in the wiki already :smiley:

Your suggestion without TX power puzzled me, because then I don't have a serial device (/dev/tty*), isn't that needed to upload at all? Isn't that needed by `dfu-util`?

benlye, 2019-05-24:
I still think it might be permissions related, the sudo test will tell us for sure.  

dfu-util doesn't use the tty device, it finds the DFU device by USB ID.  does `sudo dfu-util -l` give you any output when powering the module just with the USB cable?

benlye, 2019-05-24:
A quick way to see if the udev rules are actually working is `ls -al /dev/maple` with the device plugged in.  The rules create a symlink at that path which will only be present if things are working correctly.

boennhoff, 2019-05-24:
**Wohooooo, success!** I stumbled over several more strange behaviours (contradicting my last tests somehow), but now it (more or less) reliably let itself flash. Only method that worked so far was: module sticking on TX, TX off, USB powered, if the Dev ID is `1eaf:0004` and there actually is a device like `/dev/ttyACM0` after connecting.
What did i change/test:
- I rebooted (USB errors when doing more tests, suspended inbetween, suspect some USB problems here)
- I tested the module without being in the TX at all
- I used the failing command from the Arduino IDE prefixed with `sudo` whcih finally showed me the upload progress bar!

Just a second ago, it also worked throught Arduino IDE w/o special permissions, the first time. I am happy and sad at the same time, I just don't understand what went/is wrong. The permission problem was a reason for me to test it within Windows... I am really stuck right now.

I also tested _Flash from TX_, but it still fails. I guess that stays like that until I am able to flash the module bootloader, right?

What version do you guys normally flash? The latest tip of master?

How do you want to continue? I am actually happy now, but I am willing to help to analyze more and eventually update the Wiki for others to not have the same problems.

boennhoff, 2019-05-24:
> A quick way to see if the udev rules are actually working is `ls -al /dev/maple` with the device plugged in. The rules create a symlink at that path which will only be present if things are working correctly.

That is true, `/dev/maple` exists.

Another test without the module being in the TX succeeded also this time. Does the actual application influence the flashing mechanism? I don't get why it is so easy now...

boennhoff, 2019-05-24:
> dfu-util doesn't use the tty device, it finds the DFU device by USB ID. does `sudo dfu-util -l` give you any output when powering the module just with the USB cable?

Nope. It does return no output. Just recognized i have dfu-util v0.8 with the IDE and dfu-util v0.9 within my Arch system. Both give same result.

benlye, 2019-05-24:
> I also tested Flash from TX, but it still fails. I guess that stays like that until I am able to flash the module bootloader, right?

I'm not sure why that's not working.  The module has a bootloader (otherwise you wouldn't see the Maple USB devices), and the installed firmware has the right code for sure now that you've flashed it.  It might be something you can ask about on the ersky forums.

> What version do you guys normally flash? The latest tip of master?

Yes.

> How do you want to continue? I am actually happy now, but I am willing to help to analyze more and eventually update the Wiki for others to not have the same problems.

If you have a method that works then it's worth adding it to the documentation.  It might be worth just mentioning that you had more success with the module outside the TX.

> Another test without the module being in the TX succeeded also this time. Does the actual application influence the flashing mechanism? I don't get why it is so easy now...

It's possible that now that you've flashed it with the main branch firmware the behavior has changed because the old firmware was doing something weird (who knows what iRangeX may have done to it!).

> Nope. It does return no output. Just recognized i have dfu-util v0.8 with the IDE and dfu-util v0.9 within my Arch system. Both give same result.

I'm guess that it's running in USB mode now (`lsusb` shows the `1eaf:1004` device) so I'd expect that.  You'd need to get it into DFU mode with `upload-reset` to see it as a DFU device again.

I suspect that there was a permissions problem accessing the device under Linux, and also that with it's original firmware your module behaved unusually when powered from the TX.  If you had tried to flash from Windows with the module only connected to USB (TX powered off) it might have worked right away - it's not clear if that's a test you tried?  The Linux reboot may have kicked udev into processing the rules file properly.

Anyway, glad you got there in the end!  Feel free to close the issue.

lad1337, 2020-08-06:
i know this is old but if someone ever finds this and want to flash the irangex irx4 lite with a UART / TTY
[image]

The STM32 BOOT0 is exposed, i soldered the orange breadboard cable to it.
And connected to to VDD (3,3v)

i then flashed it with https://github.com/benlye/flash-multi (on mac via cli)

pascallanger, 2020-08-06:
It's already in the doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md

pascallanger, 2020-08-06:
And I'm not sure why you would want to flash it using serial anymore since we have a bootloader update tool using USB using Flash Multi.

lad1337, 2020-08-06:
> And I'm not sure why you would want to flash it using serial anymore since we have a bootloader update tool using USB using Flash Multi.

I had version 1.2.0.22 and did not appear as any device on my computer.

Now I see it in the docs https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#usb-to-serial-adapter
Additionally that my method might have fried the RF if I had used 5,5v.

Thanks for all the work you do!