pascallanger/DIY-Multiprotocol-TX-Module #227: XVERT will not arm

fulltimer123, 2019-05-26:
Upgraded IRangeX 4-in-1 from stock to this firmware.  No other changes, but now my XVERT will not arm.  Binds fine and all other controls work as before.  Telemetry, everything else works.

Ideas??

pascallanger, 2019-05-26:
Please read the DSM documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm---6
Extract of the part you are looking for:
> TH_KILL is a feature which is enabled on channel 15 by default (can be disabled/changed) in the _config.h file. Some models (X-Vert, Blade 230S...) require a special position to instant stop the motor(s). If the channel 15 is above -50% the throttle is untouched but if it is between -50% and -100%, the throttle output will be forced between -100% and -150%. For example, a value of -80% applied on channel 15 will instantly kill the motors on the X-Vert.

fulltimer123, 2019-05-27:
Thank you.....  I'm just learning about programming the module.  Can the fixes you mention be accomplished in OpenTX or do I need to recompile your module firmware??

closed 2019-05-31 by pascallanger