pascallanger/DIY-Multiprotocol-TX-Module #228: Flash ONLY the  bootloader to

fulltimer123, 2019-05-29:
Can ONLY the Bootloader  be flashed to IRX4-in-1; not the full firmware package.  

If so , how???

benlye, 2019-05-30:
If you flash the bootloader the rest of the firmware (the application code) will be erased, meaning your module will not work at all until you write the firmware to it. 

So yes, it is possible to flash only the bootloader but you probably don't want to do that.

fulltimer123, 2019-05-30:
Thank you benlye; I understand.  Since I will not have a copy of the stock firmware in the module I'm ordering, I will use it "as is".

benlye, 2019-05-30:
It's relatively easy to read the firmware off the module into a file to create a backup.

But really, why would you want to keep the stock firmware?  It's just an old version of this repository's code with nothing special (and missing recent updates and bug fixes).

fulltimer123, 2019-05-30:
Because flashing the latest version here, caused me major problems (Xvert would not arm).  

And because I ruined the FTDI soldering pads (my fault of course) I can't even try an older version from here.   Once bitten twice shy; as they say.  I had to order a whole new module from Bangood.

So will NEVER flash again without a backup first.

fulltimer123, 2019-05-30:
BTW, benlye.....  Any tips where I can find info on how to READ the firmware will be appreciated.  I assume avrdude would be a good way??

pascallanger, 2019-05-31:
Xvert will arm if you follow what has been said...

closed 2019-05-31 by pascallanger

benlye, 2019-05-31:
@fulltimer123 you can read the module using a serial adapter and stm32flash.  Exact method will vary depending on the model of module, and your operating system.

pascallanger, 2019-05-31:
This shows how to read/write the EEPROM which is the pretty much the same for the full firmware: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/EEPROM.md#multi-module-eeprom
I still don't understand what you are trying to do. The otiginal firmware has nothing special, it might be compiled with DSM_MAX_THROW enabled that's all. Again XVert will arm if you follow the doc at the letter: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm---6