pascallanger/DIY-Multiprotocol-TX-Module #230: Number of bind slots doesn't equal number of models possible on TX

Gfyss, 2019-06-02:
I'm trying to migrate off a Graupner MC-24 that's been my primary radio since 2004, to a FrSky X9E.  Everything I have is DSM2/X. During the transition, I discovered that I would only be able to use 16 of 64 slots on my X9E, I still have 4 aircraft on the old radio, 5 rx's for projects in work, and 7 others not assigned yet, all of them are DSM2 or DSMX. I did buy 2 FrSky rx's as a backstop, but prefer DSM's use of satellites for path diversity.

Assuming you are willing to make the adjustment and it's technically workable, my recommendation would be to increase it to the maximum number of slots allowed by OpenTX. My X9E allows 64. That would pretty much accommodate any future growth requirements for a long time.  32 would be satisfactory if 64 is not possible.

pascallanger, 2019-06-02:
It's not doable with the current serial protocol...
There is nothing preventing you to use the same RX_Num for multiple RXs. It's just that you won't have a one to one match between the model on the radio and your physical model which is kind of ok I think.
Pascal

Gfyss, 2019-06-02:
So it would behave the same as using a Spektrum conversion module on an old 72mhz system?  In other words, the only thing I'm "giving up" is the model match ability if I use the 4in1's RX_NUM1 to bind all the other receivers?  If that's true I'll go with that.  I've lived without model match for the past decade+ so not an issue for me.

Is that a correct interpretation?

Thanks!

pascallanger, 2019-06-03:
Yes that's it.
Pascal

closed 2019-06-03 by pascallanger