pascallanger/DIY-Multiprotocol-TX-Module #231: Support for Flysky AFHDS3

piotrkundu, 2019-06-03:
Some pre-req questions:
1. Is there a need for new hardware to support AFHDS3?
2a. If not, is a matter of figuring out the frequency hopping?
2b. If not, is it a matter of reverse engineering the protocol?
2c. Or both of the above.

I'm happy to get my hands dirty and help out, but I would need some guidance on the radio parts.

pascallanger, 2019-06-03:
1. No idea. Someone has to open the TX or RX to look at the RF component. So far Flysky have always used an A7105 but...
2c is correct.
One needs to open the TX locate the rf component and take a picture. From there, we can look at the feasibility to spy the SPI bus between the micro controller and the RF component to then start the reverse engineering.
The best would be to get a loaner TX and RX to work it out.
Pascal

closed 2019-06-27 by pascallanger

piotrkundu, 2019-07-05:
I will have a look when I'm finally on vacation and post pictures.

zelogik, 2024-09-04:
Hello,

I know that it's a old issue, but I try...

@pascallanger, I have two noble nb4, and some FGr4s, and many "clone" from dasmikro (not available anymore).
A working RadioMaster MT12 with 4in1 and EdgeTX could be fantastic with a AFHDS3 protocol.

Are you interested if I offer to you a noble nb4 and a original FGr4s to check/implement AFHDS3?
Ie: If it's compatible with a multiprotocol TX, you can keep TX and RX :-D
Ie2: Just find a way to send to you (ie: I'm from France)

below pictures of AFHDS3 compatible module from dasmikro:
[image]
[image]

Regards