pascallanger/DIY-Multiprotocol-TX-Module #232: Multiprotocol RX-module for USB RX dongle for simulators

piotrkundu, 2019-06-03:
I love to use my real radio for simulators and this looks like the perfect project to support multiple radios with multiprotocol-RX to HID device. Most dongles on the market have some drawbacks: 
Flyssky XSR-SIM needs to be removed and inserted after each reboot.
Most dongles are using the trainer port, so not wireless.
Most dongles are using PWM to HID solutions, so latency is in ~100ms or above

Ready to get my hands dirty, but need support on the radio parts. HW and coding is not an issue.

pascallanger, 2019-06-03:
I'm not sure what you are looking for but I think that you want to turn the multiprotocol module to a receiver, connect it to the PC through HID so it appears like a joystick. If yes this is doable and all the hrdware to do so is already in the modules iRangeX (not Jumper). It's all about coding a RX for the protocol(s) you want to support which is not too complicated since you have the source code of the TX in this github.

schwabe, 2019-06-03:
You might also look into the Betaflight repo, they already have receiver code in there.

AndreBL, 2019-06-04:
Flysky AFHDS 2A receivers with i-bus have 7.7ms output refresh interval. The receivers are relatively cheap and supported by the MPM (if your TX is not FlySky). Perhaps building a dongle that is able to convert i-bus from a FlySky RX to HID will work too.

closed 2019-06-04 by pascallanger

piotrkundu, 2019-07-05:
@AndreBL I got one of these ordered: http://rc-usb.com/rcusb-rx4ss/
@schwabe  @pascallanger Thanks, for the tips I will look into it.