pascallanger/DIY-Multiprotocol-TX-Module #235: FS-IA6B (AFHDS 2A) strange telemetry

AndreBL, 2019-06-07:
Hello.

I got a brand new FS-IA6B receiver from banggood.com and it is showing some unknown sensors in the telemetry screen of OpenTX 2.2.3. The new unrecognized sensor IDs and data as show in OpenTX screen are below:
00FD 70
00DE 65535

I noticed that this value "70" is the PWM frequency setting (I changed mine from default). Changing the PWM frequency to 65Hz made the value for ID 00FD change to 65 accordingly. As for the second ID, I have no clue about. Since 65535=0xffff, it might mean something that is empty.

Other facts:
1) The above IDs are not updated as often as known sensors. The asterisks near those values flash only once every few seconds.
2) When the asterisks for these unknown sensors flash, the value for the RX voltage sensor gets corrupted with a negative number. It is hard to notice on the screen because it happens very briefly, but if you have a logical switch taking care of alarming low RX voltage, it catches the glitch.

pascallanger, 2019-06-07:
OpenTX code is not ready to receive extended afhds2a telemetry. I'll see if I can make the extended telemetry frames becoming invalid if not I will turn it off...

schwabe, 2019-06-07:
I did not know about them at the time when I implemented the code. Is there a documentation how the extended frames look like?

pascallanger, 2019-06-07:
Hey Arne, that's been a long time!
I've sent you an email some time ago (May 12th) about the extended telemetry and also posted on the OpenTX issue but I haven't heard back on any so I was planning to work on it this WE (been really busy lately).
I think the way I've currently hacked it won't work since it's not future proof. But I don't know how to best implement it since I don't know how OpenTX exactly works in the backend to handle telemetry so why I sent you the email earlier and why I was going to look myself which is probably good anyway ;-)

There are 2 type of sensors. The one that currently OpenTX knows about are structured always the same way 4 bytes : sensor ID, sensor number, 2 bytes value. But there are other type of sensors which are sending sensor ID, sensor number, length and payload.

From a communication standpoint with the RX I know which format is currently being sent so I can pass it to OpenTX, what I don't know from an implementation perspective is if I should:
 - add a new telemetry type 0x07 (current telemetry is 0x06)?
 - add an extra signaling byte at the end of the current telementry 0x06 frame I'm sending only if it's extended telemetry.
What I'm looking for is a way to integrate without breaking current install base. Option 1 and 2 should not do anything bad since both should be ignored right?
Option1: I'm not sure what the implivations are to receive telemetry type 6, 7, 6, 7, 6, 7....

One sensor is for example the altimeter sensor:
T= 41 01 04 08 86 49 13 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
41 01 04 08 49 13 => Sensor ID: 41 -> presure sensor, Sensor #: 01, length in bytes 04, data: 08 86 49 13
Temperature= ( ((0x13498608)>>19) -400)/10 = 21.7°C

schwabe, 2019-06-07:
Hm, I did not get your email, did you send it to my normal email address? And I completely missed also the issue. Sorry about that. Don't have so much time for RC in last year unfortunately.

For the current stuff, I think using a different telemetry type is the best option.

I wonder if we should for future protocols switch to let Multi do the parsing which avoid having duplicating the code/waiting on new OpenTX releases. Kind of like we did in the past with mapping to Frsky S.port telemetry but without needing to match an existing Frsky sensor and all the uglyness that comes with that. That would also have the benefit of avoiding Frsky patent claim they currently make on the S.port protocol.

Instead we could be adding a new telemetry type like MultiTelemetry or OpenTX telemetry that allows Multi basically to directly push its values to OpenTX.

So OpenTX internally treats all sensor as having some id, some unit and some value. It matches very closely the Frsky protocol. So all the sensor parsing code maps from the raw telemetry to this tuple.

Function is

setTelemetryValue(TelemetryProtocol protocol, uint16_t id, uint8_t subId, uint8_t instance, int32_t value, uint32_t unit, uint32_t prec)

And there is another function to set the name which is 4 byte

So basically
- 32 bit id (split into id, subId, instance)
- 32 bit value
- What unit the telemetry value has (e.g. volt/mah/etc...)
- precision (like * 1/100, or 1/10), basically the number of precision digits
- 4 String name.

What do you think of that approach?

AndreBL, 2019-06-07:
I already had a bunch of crude workarounds for sensor values because I didn't want to start messing with OpenTX and Companion codes. Just added another one and my telemetry screen is now clean with the new receiver.

```
#ifdef AFHDS2A_FW_TELEMETRY
if (option & 0x80) {
    //Workarounds start here

    // 0xAA Normal telemetry, 0xAC Extended telemetry not decoded here too
    if (packet[0] != 0xAA)
        return; //This line doesn't appear to stop the unknown sensors from the new ia6b

    for (uint8_t sensor = 0; sensor < 7; sensor++) {
        uint8_t index = 9 + (4 * sensor);
        switch (packet[index]) {
        //RSNR: offset workaround for Companion allowed RSSI range
        case AFHDS2A_SENSOR_RX_SNR:
            if (packet[index + 2] > 0)
                packet[index + 2] += 20;
            break;

        //RQly: quality instead of error rate workaround
        case AFHDS2A_SENSOR_RX_ERR_RATE:
            packet[index + 2] = 100 - packet[index + 2];
            break;

        //Unknown packet from new ia6b workaround
        case 0xFD:
            return; //discard
            break;
        }
    }
    //Workarounds end here

    // forward 0xAA and 0xAC telemetry to TX, skip rx and tx id to save space
    pkt[0] = TX_RSSI;
    debug("T=");
...............
```

pascallanger, 2019-06-07:
I'm cautious of the flash size. I'm afraid that it will eat up a lot of space to decode the sensors for every protocols... It's becoming really tight on the STM32 especially when I enable debug.
Just too many protocols and I have 3 others in progress.
On the other hand, people won't have to upgrade their radios. I'm not certain what's the right thing to do here.

AndreBL, 2019-06-07:
I may be biased giving my opinion, but I'm happier with a module which has near full support for a few mainstream RC protocols with a variety of compatible receivers rather than a module with one thousand model-specific protocols for toy-grade drones, each one with special buttons and other obstacles taking the space. :)

pascallanger, 2019-06-07:
That doesn't really count here unless I'm mistaken.
If the module decodes the telemetry or the radio in the end you don't really care...
What you care is to have the information displayed on the radio. Therefore the only factor might be for you: will it be faster to get it implemented by the OpenTX team or me?

"near full support for a few mainstream RC protocols": most of the time I don't have the hardware to work and therefore wasting tons of time. For afhds2a I only have a receiver to work with and no sensors. I can't just buy every stuff to reverse it which makes it hard to add full support...

geileraupe, 2019-06-10:
HI Pascal
Here i have a voltage sensor.. If this helps i can send it to you.. I must look after a vario sensor ... if i can get some ..

pascallanger, 2019-09-19:
Issue is fixed in Multi 1.2.1.78
Full Flysky AFHDS2A telemetry will be soon in OpenTX 2.3 (hopefully 2.3.1).
Pascal

closed 2019-09-19 by pascallanger