pascallanger/DIY-Multiprotocol-TX-Module #236: unable to bind to flit10 receiver ( afhds2a)

DaSeppoFPV, 2019-06-07:
Hello 
i just got my new Jumper t12 Radio and i am not able to bind it whith my flit10 receiver via the menu.

so my question is do i have to use the ppm input from the jp4in1 Module to bind?
and if so isnt there much mode latency via ppm than i-bus?

thank you for your help

pascallanger, 2019-06-07:
PPM won't make a difference.
You need to upgrade the firmware of your Jumper 4in1 module to the latest version 1.2.1.58: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md

DaSeppoFPV, 2019-06-07:
ok so i need all the tools to upgrade the jp4in1  module? or can i Upgrade it via the usb Port?

and then i can bind whith my receiver via the open tx menu?
or do i have to bind it still via ppm and the bind button in the module?

thank you for your help

i tought the multiprotocoll is not such a hassle to set up.....

pascallanger, 2019-06-07:
"ok so i need all the tools to upgrade the jp4in1 module?"
You don't need anything to upgrade the jumper 4in1 apart from the USB cable that you've received with it as indicated in the doc...
"and then i can bind whith my receiver via the open tx menu?"
If they haven't done anything special with this receiver.
"or do i have to bind it still via ppm and the bind button in the module?"
PPM or via the menu will give the same result.
"i tought the multiprotocoll is not such a hassle to set up....."
Come on it's simple, just give it at least a try.

DaSeppoFPV, 2019-06-07:
ok thank you for your help.... 
a soon as i am at home i will give it a try.

thank you for your help an patiance 😉

DaSeppoFPV, 2019-06-07:
Thank you for your help - everything ist working fine.

closed 2019-06-07 by DaSeppoFPV