pascallanger/DIY-Multiprotocol-TX-Module #237: Xlite Pro and Vantac MPM lite

Talha909, 2019-06-11:
I have purchased the Xlite pro and Vantac MPM. I update the both devices.  I am facing one issue. My multi module did not come online when I turn on the TX. I have to switch between the models. I do this three of four time after that I saw green lite on my MPM. I need to switch an other model then I also have to switch between the models to come online.

fabioeh, 2019-06-11:
Having same Issue.

MPM Vantech module Fails to go online. Status light turns Red. Will not Bind.

Setting External RF to off and reselecting MULTI brings the MPM module online and the Status light turns green. Turning radio off and on will not bring MPM online.

Using DSM x22 protocol.

MPM ver 1.2.1.51, Opentx N21, FRSKY PMU Firmware 190605

schwabe, 2019-06-11:
Yes, basically Frsky managed to break their own module: https://github.com/opentx/opentx/issues/6481

pascallanger, 2019-06-11:
From what I understand, the module is powered but does not boot each time, correct?
If it's true then I think the issue is the bootloader of the Vantc MPM. Do you know which version it is running?
Are you able to flash the module using serial while strapping boot0? By doing so, you could try to remove the bootloader using the upload method "Upload via Serial (FTDI)" or upgrade the bootloader to the latest version using "Upload via Serial incl Bootloader (FTDI)".

Talha909, 2019-06-11:
I have yesterday flash the new version of Multi Module Multiprotocol v1.2.1.59. I flash the Module through the Usb method. The bootloader is already installed.  Please suggest me what should i do now.

Talha909, 2019-06-11:
And how i check the bootloader version.

pascallanger, 2019-06-11:
You need to use a ftdi like I've said before. Try to flash the firmware without the bootloader to see if it solves your issue.

Talha909, 2019-06-11:
Thanks .  One thing more.  I have FTDI of 3.3 volt.  can i gave power to my MPM with the ftdi. I dont wana use xliter pro. Because when I opened the MPM then its difficult to put back in xpro lite.

fabioeh, 2019-06-11:
The MPM boot loader that is part of the .59 release?

Has it been modified for the x light pro?

Is it used to power on the module.?

If so who is going to fix the boot loader ?

OpenTX said it was the MPU in the pro.

fabioeh, 2019-06-11:
With 3 companies pointing fingers how are we going to get this working.

OpenTX has no production X-light-pro to try this on. They only have prototypes that work and do not have the MPU.

fabioeh, 2019-06-11:
Looks like only customers have the production pro but not the software developers.

pascallanger, 2019-06-12:
> Thanks . One thing more. I have FTDI of 3.3 volt. can i gave power to my MPM with the ftdi. I dont wana use xliter pro. Because when I opened the MPM then its difficult to put back in xpro lite.
If you are sure that your FTDI is outputing 3.3V (check it with a voltmeter) then yes go ahead. Also make sure that you only power it while the boot0 jumper is in place otherwise I fear that your FTDI won't be able to support the current request of the module.
Use the method "Upload via Serial (FTDI)" which will get rid of the bootloader. If it works then we will move to the next step.

Talha909, 2019-06-12:
> If you are sure that your FTDI is outputing 3.3V (check it with a voltmeter) then yes go ahead. Also make sure that you only power it while the boot0 jumper is in place otherwise I fear that your [...]

One stupid question. How can i check that its work or not.  I will flash it with v1.2.1.59.

pascallanger, 2019-06-12:
Latest version is 1.2.1.61, try to upload this one with the method "Upload via Serial (FTDI)". If you see the new version then you know you flashed it correctly and if the correct method has been used the bootloader is gone which means that USB won't work anymore when you plug it to the PC which will appear as an unknown device.
Yes you'll be able to install the bootloader again, we are troubleshooting here.

Talha909, 2019-06-12:
I did not see the latest firmware. I only see the v1.2.1.59. How can i download the latest firmware.

pascallanger, 2019-06-12:
> The MPM boot loader that is part of the .59 release?

When you flash v1.2.1.xx you only flash the firmware not the bootloader. The bootloader is only flashable by using a FTDI.

> Has it been modified for the x light pro?

No, I'm trying to troubleshoot the issue. I've never seen a x-lite pro, no idea what they've done to the power on this remote, no idea what bootloader firmware they are shipping with the Vantac multimodule since I don't even have one and so on...
 
> Is it used to power on the module.?

The bootloader is the first thing executed, it could wrongly see commands at power up on the serial pin putting it in programmation mode instead of booting.

> If so who is going to fix the boot loader ?

Anyone who wants to particpate, the bootloader is open source: https://github.com/MikeBland/StmMultiBoot

> OpenTX said it was the MPU in the pro.

Yes and is that helping you?

> With 3 companies pointing fingers how are we going to get this working.

I'm not sure which companies you are talking about... There is only one company here FrSky producing and selling the transmitter and module hardware. For the multimodule, I'm the really the main dev doing this on spare time and not sponsored in any form. This is why you can see on my github page a donnation button: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Donations.md . FrSky, iRangeX and Jumper are all selling the hardware with our software on it without giving in return.

> OpenTX has no production X-light-pro to try this on. They only have prototypes that work and do not have the MPU.

I have no link with FrSky in any form so not receiving any product to test from them. And guess who needs to take care of their customers?

> Looks like only customers have the production pro but not the software developers.

We have to purchase the hardware like you and guess what I just can't afford it!!!
Same goes with all the models, a lot of people are asking for new protocols but how to add a protocol without having the hardware and when you ask for something you don't even get an answer...

Pascal

pascallanger, 2019-06-12:
> I did not see the latest firmware. I only see the v1.2.1.59. How can i download the latest firmware.

Follow the standard process:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#download-the-multiprotocol-source-and-open-the-project

Talha909, 2019-06-12:
XLITE PRO ON OPEN TX N20 FIRMWARE
MPM Lite on v1.2.1.60

Today I fond one interesting ting.  I have installed two models. One is D8 and other one is D16.  When switch on D16 mode then MPM comes online .  But when I switch to D8 mode. The green light and red light start continues. I switch between models 5 times from D16 to D8 . After 5th try its comes online on D8 Mode. But from D8 to D16 its comes online on first time. I repeat the same process 5 times and result was same. From D16 TO D8 its takes 5 or 6 time switching and from D8 to D16 Its comes on first try or in 2nd try.

pascallanger, 2019-06-12:
Have you tried without the bootloader?

Talha909, 2019-06-12:
Not yet. I will go to market in evening and solder the pins from the person.  then i will update it .  I am not good in soldering.

Talha909, 2019-06-12:
I have try the update the MPM withe ftdi and its works fine. Now please tell me how can i update the bootloader

Talha909, 2019-06-12:
thanks man. Its works now.

pascallanger, 2019-06-12:
To write back the bootloader and firmware just use the option serial including bootloader.

Talha909, 2019-06-12:
thanks man.

pascallanger, 2019-06-12:
You can always donate a little something if you are happy 😉

closed 2019-06-12 by pascallanger

CDM040971, 2019-08-31:
Hi Folks, 
The solution above is not clear to me. I have the issue described in this article. 
Running MPM Lite Module
Tried with both firmware versions: 
-> Multi-STM_TXFLASH_INV_OPENTX-v1.2.1.71.bin
-> Multi-STM_TXFLASH_INV-v1.2.1.71.bin
Still on radio boot up with module selected with Internal RF Disabled and External RF MPM Lite Enabled - Red Light - No Telemetry - startup always fails. 

To activate MPM Lite RF I do this:
Select a Model with Internal RF Enabled - done - then go back and switch to a Model with External RF Enabled and the MPM Lite activates - green light!

I have Taranis x9-Lite Radio with OpenTX 2.3 Night ver N73

Please let me know how to fix the issue so that at boot up, the External RF - MPM Lite module activates.

pascallanger, 2019-08-31:
You need to update the bootloader to the standard one. I don't know which bootloader FrSky is shipping but it does not work correctly.
To do this you need to open the module and use a 3.3V FTDI to flash the module.

Talha909, 2019-08-31:
> Hi Folks,
> The solution above is not clear to me. I have the issue described in this article.
> Running MPM Lite Module
> Tried with both firmware versions:
> -> Multi-STM_TXFLASH_INV_OPENTX-v1.2.1.71.bin [...]

Its very easy. I have the same issue but issue was resolved with the help of pass. do you have facebook Id. I can help you for the flash of the new firmware.

CDM040971, 2019-08-31:
Hi Talha909, My FB ID is at https://www.facebook.com/chrisdmeyer or email via ChrisDMeyer@gmail.com.

Thank you kindly. 
Chris

CDM040971, 2019-09-01:
Hi Talha, 

I have found I need to switch models exactly 5 times before the green light shows. So frustrating. I also don’t have the 3.3v flash unit. So hopefully your solution works via direct USB connection. Cheers. Txs. Chris.

pascallanger, 2019-09-01:
USB connection can't flash the bootloader. You need a 3.3V FTDI. it's only a couple of $....
Pascal

CDM040971, 2019-09-01:
Hi Pascal. 

Will this product work for what I need to do. 
https://buzzfpv.com.au/ftdi-5v-micro-usb-turn-to-ttl-mwc-cp2104-debugger-programming-unit?search=Ftdi&description=true

pascallanger, 2019-09-01:
No you need a FTDI which would work in 3.3V, this one $1.68 is what you are looking for: https://www.banggood.com/FT232RL-FTDI-USB-To-TTL-Serial-Converter-Adapter-Module-For-Arduino-p-917226.html?rmmds=search&cur_warehouse=CN

Talha909, 2019-09-01:
i have added you in facebook. but you r not on line

CDM040971, 2019-09-01:
Thank you Pascal.

CDM040971, 2019-09-01:
Hi Talha. I live in Perth Australia. Your message came through at midnight 😀 I can be available at the time of this message tomorrow? Which is 6am here. As Pascal has mentioned I’ll be ordering the TDFI board from Banggood which takes 25 days for delivery.

pascallanger, 2019-09-02:
It was only an example, you can use other sources which might be faster but you need to make sure that it can do 3.3V

CDM040971, 2019-09-02:
Thanks Pascal. As soon as I get it I’ll reach out. 👍😎

CDM040971, 2019-09-09:
Hi Pascal, 

I received this module.... can this work? Its a ST Link V2 STM8/STM32 programming unit.

https://core-electronics.com.au/st-link-stm8-stm32-v2-programmer-emulator.html?utm_source=google_shopping&gclid=CjwKCAjw8NfrBRA7EiwAfiVJpdTf4vbBNyaV1gDP_dvxejWJwLysesNOJ04GWBcsPqBvIXVnNLJo4RoCavsQAvD_BwE

Thank you
Chris

pascallanger, 2019-09-10:
No it won't work... You need a standard USB to serial adapter (FTDI).Pascal

CDM040971, 2019-09-10:
Hi Pascal, Talha,
Your guidance has helped me be successful and now my MPM Lite module works every time.  
I have captured some working notes and shared them here to help anyone else. 
With your approval Pascal, if not approved. Please delete my post. 

Thank you kindly. 
[Avantac MPM Lite Red LED Issue.docx](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3594841/Avantac.MPM.Lite.Red.LED.Issue.docx)