pascallanger/DIY-Multiprotocol-TX-Module #238: FrSky L9R receiver support (LR12 mode)

SilverFire, 2019-06-16:
Hello, team!
Thank you for making this amazing anti-vendor-lock project!

I have a few [FrSky L9R](https://www.banggood.com/FrSky-L9R-Long-Range-RC-Receiver-for-FrSky-Taranis-X9D-p-935605.html) long range receivers, that work on 2.4 GHz with LR12 protocol, originally supported by [XJT module](https://www.banggood.com/FrSky-XJT-2_4Ghz-Combo-Pack-for-JR-w-Telemetry-Module-With-X8R-Receiver-p-1060775.html).

I would like to see this protocol support implemented and will help you to the best of my possibilities. Could you check this protocol, please?

pascallanger, 2019-06-20:
I've got neither the TX or RX to work with so it will be difficult to add support...

SilverFire, 2019-06-20:
I have RX and can borrow TX, but how can it help you?)

pascallanger, 2019-06-20:
I need multiple dump of the SPI bus from the RF part of the TX. Captures of the TX bind, channels moving...

closed 2019-06-27 by pascallanger

Talha909, 2019-10-11:
How we get the multiple dump of the SPI bus from the RF part of TX

pascallanger, 2019-10-11:
You need to open the TX, locate the RF component CC2500. Solder wires on the SPI bus between the MCU and CC2500. Connect a logic analyzer and launch some dumps.

pascallanger, 2019-10-11:
If you feel like doing it let me know.