pascallanger/DIY-Multiprotocol-TX-Module #239: Useful? FRSKY ACCESS Protocol Documentation

justingoesfpv, 2019-06-20:
Not sure if you can find any use in this, I couldn't find it "publicly" available anywhere.

[ACCESS protocol.docx](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3308304/ACCESS.protocol.docx)

SilverFire, 2019-06-20:
Wow, thank you!
Maybe you have the same for LR12 protocol? Looking for it for https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/238

justingoesfpv, 2019-06-20:
Unfortunately I do not. They have a "public" server. It has the ACCESS Docs, some BLE5 bootloaders, BLE5 app, and the freelink app(source code) on it. Looks like it also has FrOS and some other goodies, but organization permissions levels do not allow me to see anything but the names.

If I come across anything else in my research pertaining to FrSky protocols, i'll drop them somewhere here!

OpenUAS, 2019-07-22:
@justingoesfpv, nice :+1:  more stuff, links, protocol info appreciated...

pascallanger, 2019-10-16:
I've saved the doc and closing this issue for now.

closed 2019-10-16 by pascallanger

andika207, 2022-02-03:
@pascallanger any news about ACCESS ?