pascallanger/DIY-Multiprotocol-TX-Module #244: Bank ID not stored in EEPROM in STM32 PPM mode

OpenUAS, 2019-07-22:
Using module iRangex 4in1 Plus, setting the bank with bind button via module yellow knob on pos 15, one sees the LED blinking 1, twice till 5 if bind is pressed. After rebood always first bank.

Using Arduino IDE 1.8.5 and 1.8.9, code update of boar to 1.1.4 make all no difference. using GCC 4.8.x (2014) official under Ubuntu GNU/Linux 16.04 64bit

Having the Serial debug cable connected to module to investigate gives in terminal:

```
Multiprotocol version: 1.2.1.67
Protocol selection switch reads as 6
Using bank 0
Generated ID from STM32 UUID
...
```
etc.etc.

Added some extra debugging code, the read bank result was always 0xFF en the code then resets the bank to 0 again. 

Not the only one with this issue looking at forums and some youtube comments read somewhere.

Still investigation to see where it goes wrong, running through full source. E.g. maybe EE_ADDR uint16 on STM32 and not uint8_t and then sdome bitshifting is messed up maybe...

Have all TX modules in existence to debug more, JTAG debug, logic analyzer, the work but rather not go that way ATM...sigh... just wanted to revive two MX22 TX with PPM and add some protocol. If debug needs, have dozens of various TX's to test on, old PPM till new Taranis X9D etc.
It just came bit as a nasty surprise, to use it and basic DSMX binding even not to work. Well... will see to find some time to debug more, in the mean time a hint welcomed.

For now the personal dirty workaround is to set the protocols needed on bank 1 via specific crafted `_MyConfig.h`

pascallanger, 2019-07-22:
Hummm, I'm on vacation right now so I can't test but the output above shows that even the ID is being regenerated where it should be done only once for all.
Have you tried to look at the "boot" output without reflashing the module? The goal is to see if it does generate a new ID again just when you powercycle it.
The thing I'm wondering is that if when you flash the module the full flash is erades then the eeprom content is lost.
Pascal

benlye, 2019-07-22:
I just checked the 64-bit Linux upload scripts.  For some reason they don't match the 32-bit scripts, and don't include changes I made some time ago to preserve the EEPROM during a flash. Either I never made the changes there, or they got overwritten at some point.  I'll get that fixed.

@OpenUAS, can you try copying `serial_upload` and `serial_upload_inc_bootloader` from `~/.arduino15/packages/multi4in1/hardware/STM32F1/1.1.4/tools/linux` to `~/.arduino15/packages/multi4in1/hardware/STM32F1/1.1.4/tools/linux64` ?

(The path may be different depending on your installation).

The scripts from the `linux` folder will preserve the last 2KB of flash space when flashing, thus preserving the EEPROM.  This may not solve the original problem, but will help to debug as your EEPROM won't get erased every time you flash the module.

OpenUAS, 2019-07-23:
@pascallanger enjoy the Holiday, mighty fast response though, thx. And yes at boot the ID get regenerated. So likely also no storage there. 
@benlye thanks for the hint will try and add results here...

pascallanger, 2019-07-23:
I'm really puzzled... If that was the case I would have thousands of reports since some models must have some data stored in the eeprom or they won't work at all.
Can you dump your eeprom after flashing the module and after a power cycle? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Topics.md#eeprom-backup-and-restore
Pascal

OpenUAS, 2019-07-23:
No luck. copied upload scripts to linux64,..
Compiled, flashed via FTDI ....powerdown TX, switch set to 15 to set bank, for example set to bank 2..., Powerdown, Switch to 6, just a random protocol one... Restarted once more to see if ID get regenerated and is regenerated AFAIK since the output is like:

```
Multiprotocol version: 1.2.1.67
Protocol selection switch reads as 15
Using bank 0
Using bank 1
Using bank 2
Multiprotocol version: 1.2.1.67
Protocol selection switch reads as 6
Using bank 0
Generated ID from STM32 UUID
Module Id: 6862ee3f
Protocol selected: 28, sub proto 0, rxnum 2, option 0
Init complete
Multiprotocol version: 1.2.1.67
Protocol selection switch reads as 6
Using bank 0
Generated ID from STM32 UUID
Module Id: 6862ee3f
Protocol selected: 28, sub proto 0, rxnum 2, option 0
Init complete
```

Will debug more in a few days, but need testflying with the specific "dirtyfix", only the needed protocols in first bank. Also setup with a fixed TX ID since 2x the same for UAS EVLOS (Extended Visual Line of Sight) handover flightstest.

OpenUAS, 2019-07-23:
@pascallanger previous post was still being typed..., before I've read the post before 3 minute difference ;) Will see to dump the eeprom

benlye, 2019-07-23:
I double-checked in my 64-bit Linux VM and it looks like the scripts get called from the /linux/ folder, but the binaries get called in /linux64/, so the scripts being out of sync wasn't the problem - the EEPROM will be getting preserved through flash cycles.

I also checked my development board and I do **not** see the module ID getting regenerated every boot.  I did a full erase (including EEPROM) then flashed via serial with SERIAL_DEBUG.  I got the UID generated on the first boot, then it gets read from EEPROM after that.

@OpenUAS, I wonder if your EEPROM data may be corrupt.  You could try erasing the entire board with stm32flash (which is in the linux64 tools folder):
`stm32flash -o /dev/[your serial port]`

Then re-upload the firmware using the IDE.

OpenUAS, 2019-07-23:
@benlye Ran EEPROM test app, that worked, just to be on safe side erased with -o... 
Bank select works now :tada: . 
So somehow EEPROM got corrupted and would be the likely cause.

@pascallanger @benlye Thanks to your hints can now get onto some overdue pressing task now.. :fireworks:

closed 2019-07-23 by OpenUAS

pascallanger, 2019-07-23:
Too bad... It would have been great to see the content of the eeprom section which could have helped to understand what's going on and may be implemented a work around.
BTW what's the eeprom test app you are mentioning?
Pascal