pascallanger/DIY-Multiprotocol-TX-Module #245: irangex xr4plus loses telemetry after FW update

perkinbr, 2019-07-24:
I hope this isn't a FAQ, but I couldn't find an explanation for this.
I have an x9D+ with OpenTX 2.2.4. 

I bought an irangex xr4plus from banggood.  I was playing around with it, and I could see the module FW version in the module status line, which was 1.2.0.22 and it had the bootloader installed.

I was having some problems with binding one of my cheap drones so  I decided to update using precompiled Multi-STM_TXFLASH_INV_OPENTX-v1.2.1.67

After that point  I could no longer can see module status, I just see No MULTI_TELEMETRY.  I made no other changes to configs or anything (that I'm aware of).
I can still control my models ( I tried one DSMX and one Symax) , though, so the module seems to be working.  The Green and Red lights are both on though the red light blinks (~1 second on ~.1 seconds off).  I don't have a copy of the original FW so I can't roll back to the original.

I also tried some earlier versions FW versions but had the same results.

benlye, 2019-07-25:
Might be a dumb question, but have you power-cycled the radio after flashing, before looking for the status?

I was just doing some testing and noticed that the first time I powered up the radio after flashing I got `No MULTI_TELEMETRY`.  After scratching my head a bit I turned the radio back on (I had turned it off while I checked some other things) and the module version was displayed.

For what it's worth, I flashed the Multi-STM_TXFLASH_INV_OPENTX-v1.2.1.67.bin file to the JP4IN1 in my T16 (yes, not the same module or radio) and I get telemetry, so whatever's going on, it doesn't seem to be a problem with the pre-compiled firmware.

LED meanings are here:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Troubleshooting.md#red-led-bind-led

Your module is waiting for a bind.

perkinbr, 2019-07-25:
Thanks. I did power cycle a couple times.

I'm wondering if I might have a hardware problem. 

I'm thinking of trying to put a uart or a logic analyzer on it, but I'm not sure what I should expect to see.

perkinbr, 2019-07-26:
Hardware issue.  Looks like in the process of upgrading the module I bent the telemetry pin.
Bending it make fixed the issue **embarrassed face.**

closed 2019-07-26 by perkinbr