pascallanger/DIY-Multiprotocol-TX-Module #246: Frsky x-lite remote control uses AFHDS2A protocol, telemetry voltage problem

Rayenchen2353, 2019-07-26:
After I installed the MPM lite High frequency head on the frsky x-lite remote control（opentx2.2.3）, after using the AFHDS2A Binding receiver as shown in the attached figure, the sensor found three data, TRSS, A1 and RQly. I found that A1 is Indicates the receiver battery voltage. My receiver is a 1S battery directly connected to the receiver. It always shows the voltage of 5.11V, and it will not change. I originally thought that the voltage is the operating voltage after voltage regulation, but I am in the flysky i6X remote control. The return voltage received by the receiver on the frequency is the normal 1S battery voltage.

Attached pictures are stored in the google cloud disk：
https://drive.google.com/open?id=1lOqqR0hN8fGwfmONiSyGIthzgi5D3kml
https://drive.google.com/open?id=1OQikxO12sKafPlLzov5OMAO-Pu5b4dJX
https://drive.google.com/open?id=1sivGaDpjZ2oMzBMEvcvuI4HQ7XBO25I7

Why does the voltage of the telemetry return does not change?

pascallanger, 2019-07-26:
Your module version is quite old, could you update it to the latest version: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md
You are confirming that you are not just looking at the voltage value but you tried to charge the lipo to see if the value is decreasing? I'm saying this since the voltage value needs to be calibrated on OpenTX to display it correctly.

Rayenchen2353, 2019-07-26:
Because the voltage value has not changed
Even when pushing the throttle and the motor load is 100%, the voltage value has not changed. I think even if there is no calibration, the voltage will not change.
And using a voltmeter to check the battery voltage before and after using it for a period of time is definitely a change

I will try to upgrade the version and still exist when viewing the problem.

Rayenchen2353, 2019-07-27:
After upgrading to v1.2.1.67, the telemetry voltage value A1 still does not match the actual voltage.

pascallanger, 2019-09-20:
Can you flash Multi v1.2.1.79 in debug mode and paste here the telemetry packets the module is receiving? [How to debug](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md)
Pascal

qba667, 2019-09-24:
@Rayenchen2353 What receiver are you using? Just cross verified with latest multi-protocol module firmware and OpenTx 2.3 all good with combination with IA6B.

Rayenchen2353, 2019-09-25:
The receiver used is [this,](https://www.banggood.com/Mini-Flysky-AFHDA-Compatible-8CH-Receiver-PPM-Output-With-1mm-JST-Socket-For-DIY-Micro-Quadcopter-p-1097424.html?rmmds=detail-left-hotproducts__1) flysky's second generation remote control protocol AFHDS 2A.
I will update the latest multi-protocol module firmware tonight, and then observe if the acquired sensor information is correct.

pascallanger, 2019-09-25:
@Rayenchen2353 You need to go in debug mode. We need to see what the telemetry frames are like I've requested previously following this procedure: [How to debug](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md) . It's really simple and we will see what the RX says.

Thanks for the link I was not aware this small receiver did exist ;-) It looks exactly the same as [this one](https://www.banggood.com/2_4G-4CH-Micro-Low-Voltage-DSM2-DSMX-Compatible-Receiver-Built-in-Brushed-ESC-p-1072251.html?rmmds=search&cur_warehouse=CN) which is in DSM2. The problem of the DSM2 one is that the PWM is starting quite high so you cannot do really slow flyers with it or you can but you are laways at the limit between motor off and on... Have they corrected that on this RX?

Rayenchen2353, 2019-09-25:
I can't update the debug feature. Follow the image options and click Upload. The error message is displayed. [图片1](https://drive.google.com/open?id=1up-NcsOTc2VHh54_TMSWnq8pTCWa6YqZ)
  However, if the Debug Option selects Native USB Debugging, the upload is successful, the TX and RX are paired, and the Serial Monitor window is opened to refresh the same content, as shown in the figure.[图片2](https://drive.google.com/open?id=13XjF9bTJGXvReIzTXq3YAe5CVnV_QUyr)
Telemetry sensor status [图片3](https://drive.google.com/open?id=1dLNO13Hvdxq3M1S_eXD9j_caKDivxCgC)
Multi-protocol version 1.2.1.81
Opentx version 2.2.4

Rayenchen2353, 2019-09-25:
> Thanks for the link I was not aware this small receiver did exist ;-) It looks exactly the same as [this [...]

No correction，this small receiver still exists

Rayenchen2353, 2019-09-25:
I tried to use another receiver with telemetry, DSMX protocol, and found that Serial Monitor has no other output. Of course, the telemetry voltage of this receiver is correct.
[AR3201G](https://drive.google.com/open?id=1zsH0Qvr9LXs2Z44P3_zH4h_dH63sQIu6)

qba667, 2019-09-25:
@Rayenchen2353 telemetry log clearly shows that you are getting value 0x01 0xff that is exactly 511 in decimal system for the sensor 0 (Internal voltage A1). I would say it is receiver bug.

pascallanger, 2019-09-25:
@Rayenchen2353
Your sensor display on OpenTX looks really weird. Can you delete all sensors and launch a new discovery?
The telemetry frames on the screenshot indicate that the RX is sending voltage and error rate. The voltage reported is 0x01FF => 5.11V so OpenTX is displaying the right value... Can you confirm that this 5.11V value is not moving at all even if you change the lipo or put it under load?
Instead of a screenshot of the debug output, can you leave the debug running for a long time and select the full text to past it in a file?
Do you see something else appearing time to time or always the same lines "00 00 FF 01 FE 00 00 00 FF FF ...."? You need to have "Autoscroll" at the bottom left checked to see it scrolling.
If you don't see anything else, we will need to remove the filters on the incoming telemetry frames to  see what's going on...
Pascal

pascallanger, 2019-09-25:
@qba667 The weird thing is that his Flysky i6X is showing the correct lipo battery voltage (see his first post).

qba667, 2019-09-25:
@pascallanger  this can not be because there is also temperature sensor there - this RX is not supporting external sensors so simply display on I6X is not with this RX.

Rayenchen2353, 2019-09-26:
> @Rayenchen2353
> Your sensor display on OpenTX looks really weird. Can you delete all sensors and launch a new discovery?

 Remove all sensors and start a new discovery, the sensor values are still the same as before

> The telemetry frames on the screenshot indicate that the RX is sending voltage and error rate. The voltage reported is 0x01FF => 5.11V so OpenTX is displaying the right value... Can you confirm that this 5.11V value is not moving at all even if you change the lipo or put it under load?

  Yes, the voltage value of 5.11 does not change even under load.

> Instead of a screenshot of the debug output, can you leave the debug running for a long time and select the full text to past it in a file?
> Do you see something else appearing time to time or always [...]

  I observed that all the output has not changed, it takes about 5 minutes.

> If you don't see anything else, we will need to remove the filters on the incoming telemetry frames to see what's going on...

Thanks Pascal and  qba667 

  it seems that the problem with the receiver is relatively large, and I may have to return it to the factory for repair. The remote control in the first post is the test chart given by the store.

closed 2019-09-26 by Rayenchen2353