pascallanger/DIY-Multiprotocol-TX-Module #247: Disabeling the configuration if OrangeTX to PPM incorrect

OpenUAS, 2019-07-26:
In validate.h there is written:

```
...
//Change/Force configuration if OrangeTX
#ifdef ORANGE_TX
	#undef ENABLE_PPM			// Disable PPM for OrangeTX module
...
```

Should this not be
`#undef ENABLE_SERIAL		// Disable SERIAL for OrangeTX module`

This module only has a PPM input. Just to be very sure, defining ORANGE_TX we talk about this module containing a Blue or Green PCB right?

[image]

pascallanger, 2019-07-26:
This is the right module.
Serial is working for sure so validate.h is correct.
I'm not sure about PPM, this needs to be tested as I don't have a module. Are you willing to give it a go?
Pascal

pascallanger, 2019-07-26:
In fact I remember now, PPM is disabled/untested since there is no hardware way to select which protocol is being launched.
So if you really want to use it with PPM you also need to force the bank and the correct ppm entry in the table and test to see if it works.
Serial gives you access to any of the CYRF protocols along with all the options which is how it has been developped as a goal.

OpenUAS, 2019-07-27:
Did not know it had serial, will test PPM mode then Hava blue and green version...

OpenUAS, 2019-07-31:
Move to own issue queue of Fork as not to pollute this tracker to much. When fixed and tested and all works 100% a new PR will be created.

closed 2019-07-31 by OpenUAS