pascallanger/DIY-Multiprotocol-TX-Module #250: Attitude TGY-CAT01sensor do not work

Djodjotje, 2019-08-11:
i use the Xlite and the irange module. 
I bind with a Flysky FS-1A10 with a FS-CVT01 voltage sensor and a TGY-CAT01 attitude sensor. The telemetry works fine exept the attidude sensor.
 I can read something on 0041 and 008B but that has no sense. With my old Flysky transmitter there is no problem.

What did i have to do to make it working as a Frsky vario...?

pascallanger, 2019-08-11:
The altitude sensor is not supported yet.
What do you have to do? If I had the sensor that would give me some push to get it done...

Djodjotje, 2019-08-11:
Oke.... i will do an donation of $ 10,===

OpenUAS, 2019-08-13:
@Djodjotje would you be so kind as to adjust your issue headerline from "**Attitude** TGY-CAT01sensor do not work" to "**Altitude** TGY-CAT01 sensor does not work"? TIA

pascallanger, 2019-09-17:
Never received anything from you... I will try to find someone with this sensor to test out my new code.
Pascal

closed 2019-09-17 by pascallanger

Djodjotje, 2019-09-17:
Hello Pascal,

 

What did you expect to received.

 

Erik

pascallanger, 2019-09-17:
The new multi code is out to take care of this sensor and many others. I've worked on the OpenTX part which is currently being tested by someone. A PR will be made in OpenTX 2.3 when tested ok.
Pascal

Djodjotje, 2019-09-18:
Pascal .

 

I tested it with a Frsky X-Lite with open TX 2.3 RC04 and with the iRX4lite Multiprotocol v1.2.1.76. With the receivers FS-iA10 and the FS iA 6B with the sensors CVT 01 and CAT 01.

 

The voltage works fine in all cases and is also recognized as A3, the altitude comes in at 0041 and is displayed as a number somewhere around 48000, the latter changes quickly and is not as stable as the "Frsky vario."

 

In the sensor tab I have to set the "ratio" to 0.1 and give a telemetry reset then it works fine ............. BUT .... THE WRONG SIDE ON ....

 

if I set the vario then he will beep when lowering and buzzing when rising, if I go UP two flights of stairs he will indicate negative - \ - 6 meters.

 

I think it is easy to solve.

 

Erik

Djodjotje, 2019-09-18:
[image]

This

pascallanger, 2019-09-18:
The changes are not in OpenTX 2.3rc04... I've said " A Pull Request will be made in OpenTX 2.3 when tested ok." So it means that it is still in my dev branch.
If you know how to compile OpenTX by yourself then you can use my branch: https://github.com/pascallanger/opentx/tree/sweet-16-squash
Otherwise you'll have to wait that we finalise the tests before it is available.
Pascal

pascallanger, 2019-09-20:
The altitude sensor is now working. It needs both Multi v1.2.1.79 and my OpenTX branch https://github.com/pascallanger/opentx/commits/sweet-pascal
A PR will be made when release candidates will be done as no new features are currently allowed to be incorporated in OpenTX.
Pascal

Djodjotje, 2019-09-22:
Hello Pascal,

 

In v1.2.1.76 there is a sensor 0041 02 and that is the vario but in v1.2.1.81 the sensor is gone ….

 

I want to try your sweet Pascal but I can not find a way to flash it on the radio.

 

Thanks,

 

Erik

pascallanger, 2019-09-23:
This is normal. This sensor is sent by another mean that your OpenTX version does not understand. You need either to compile by yourself or wait that it gets in nightlies.
Pascal