pascallanger/DIY-Multiprotocol-TX-Module #252: Taranis X9-Lite and Vantac MPM lite - No "Multi" Option

CDM040971, 2019-08-27:
Hi Folks, 

Wondering if someone has a fix for the error I'm seeing. Received my new "Vantac MPM lite". Disabled internal RF.
Enabled external RF - No "multi" protocol selectable option. 

I have followed the "pascallanger" firmware install guide - Awesome Work Mr Pascal - Thank you

Downloaded FW : Multi-STM_TXFLASH_INV_OPENTX-v1.2.1.71.bin 
Using FlashMulti connecting the MPM directly via USB port on minim module. 
The firmware loads fine.

After all that plug the module in and nothing has changed I still don't see "MULTI" as an option.

pascallanger, 2019-08-27:
Your opentx version does not have the support for multi. You have to select it in companion and upload the firmware to your radio.

CDM040971, 2019-08-27:
Hi Pascal.

I’m running the latest OpenTX nightly Build to provide ACCST D16 support.

Do I still need to update in companion?

Thanks again
Chris

pascallanger, 2019-08-27:
> Hi Pascal.
> Decided to revert back to original firmware, unfortunately I cannot select my radio type in Companion 2.2 meaning I cannot go back and need to stay on nightly build 2.3.
> I guess this means I'm stuck waiting on an update in the nightly opentx build.

You should open an issue in the opentx side. Nothing I can do to help...

CDM040971, 2019-08-27:
[image]

[image]

Companion does not seem to have the required multi firmware support.

CDM040971, 2019-08-27:
Hi Pascal. 
Decided to revert back to original firmware, unfortunately I cannot select my radio type in Companion 2.2 meaning I cannot go back and need to stay on nightly build 2.3. 

I guess this means I'm stuck waiting on an update in the nightly opentx build.

CDM040971, 2019-08-27:
Thank you Pascal

closed 2019-08-27 by CDM040971