pascallanger/DIY-Multiprotocol-TX-Module #271: Intermittent issue using the URUAV TMX5 Lite using FrSky protocol

CurryKitten, 2019-09-30:
I've got the (relatively new) URUAV TMX5 Lite module which runs all 4 of the supported chips using an STM32.  I didn't see it in the docs, so don't know if it's "officially" supported.

Support for this module was provided for the newer FrSky radios such as the X9-Lite pro in OpenTX 2.3, or which a stable version is now out, although it's supported on older 2.2 versions of OpenTX through radios like the X-lite.

The problem I'm seeing is that when sometimes attempting to use the FrSky protocol, regardless of subtype, more than half of the time, OpenTX receives what it deems to be an invalid status message and so lists the status as "No Telemetry" when it does initialise properly, this is replaced by the version of the modules firmware that's running.  The module shipped with V1.2.0.20.  I saw this was quite old and so flashed it with v1.2.1.84.  The same issues still occur though.

I tried using the debug option in the compilation to see if I could capture the issue and see where things were going adrift, using the serial USB connection, but in every case, the module behaves absolutely perfectly when plugged in via USB, and so we just see the regular debug info -

`Multiprotocol version: 1.2.1.84
Protocol selection switch reads as 0
Using bank 0
Read ID from EEPROM
Module Id: 7d408c69
Init complete
Protocol selected: 3, sub proto 0, rxnum 1, option 0
`

The fact it works like this makes me think it's a timing/initialization issue.  When we are stuck in this "no telemetry state" I am able to go to the Hubsan protocol and get a status from the module there, albeit using a different RF chip.

When the problem is seen, both led's are solid, and attempting to bind does nothing.  When it works, I can just see a single green led, and red will flash correctly if you attempt to bind

pascallanger, 2019-09-30:
I've never seen this module before and I wasn't aware it existed... If you could do some good pictures of the internals I would be interested.
It's strange that when the USB is plugged it works correctly. Have you tried to not put the module in debug but just plug it to a USB charger for example? Plugging it to USB will provide power to the module all the time, which might give us some indications if it doesn't hang that way.
Also is it happening when you change the protocol or when you boot the radio with Frsky already selected?
Pascal

CurryKitten, 2019-09-30:
I'm attaching some images of the internals (open module case and the reverse side of the main board)  Hopefully they are of decent enough quality.

[image]
[image]

So if I recompile the code without debug active and then power the module just by a USB power bank, it appears to boot and then stop.  So rapid red flashing light, followed by a steady 1 second red led blink for approx 11 seconds after which point it shuts down. 

Presumably, at this point it's waiting for comms from the host radio.  So I wouldn't necessarily call it a hang as such.  Certainly it seems to stop responding for the given protocol, but if I went from the non-responsive FrSky protocol, to Hubasn for example, it would react properly and flash the correct leds.

Whether I power up with FrSky chosen or move to a model using that after boot doesn't seem to make a difference.  Most of the time it shows "no telemetry" and sometimes it works and shows the version number.  I've noticed at if I'm willing to move back and forth between a the FrSky model and another model it will eventually seem initialize correctly and be usable. 

It certainly seems curious that simply having serial debug running would cause the module to work everytime.  Obviously it's booting from USB power alone, but (if I'm reading the code correctly) it doesn't initialise the FrSky protocol until the protocol is selected - at which point the module would generally be in the same situation whether it was powered up from USB or the radio... unless there's some subtle timing change because of the serial writes ?

pascallanger, 2019-09-30:
I don't think the debug is what's making the module to work. This is why I was more thinking of a power issue. You 've tried to power it on from USB but it looks like you haven't powered on the radio?
What happens to the LEDs when you switch from one protocol to another?

CurryKitten, 2019-09-30:
I've been using the radio, but when running with debug, it did take the power from the USB.  Ok, so interesting about swapping protocols.  If I power up in FrSky and it doesn't work (status is "no telemetry" and both leds are on. If I cycle through the protocols within the model, there's no change the led's stay lit, and the status lines (when reported) carry on saying "no telemetry"  It's only when I swap to a different model on OpenTX that something will happen.  I don't know it OpenTX reboots the module?  Anyway, if I boot up on a working protocol, then changing the protocol within that model will also work everytime, FrSky included, and the leds seem to respond correctly, so you see a red flash with the autobinding protocols and just a solid green for others

pascallanger, 2019-09-30:
I wish I could reproduce it. Which version of OpenTX are you running?

pascallanger, 2019-09-30:
There is a known bug on this module which is the same as the Vantac module but I'm not it is linked. The problem is in the bootloader they are using. I don't know which version they are using but it's a bad one. Would you be able to upgrade it?
Someone made a doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/3594850/Vantac.MPM.Lite.Red.LED.Issue.docx

CurryKitten, 2019-09-30:
I'll find my FTDI interface, locate a windows machine and give it a go.  looks a bit involved, but yes I'll do that and report back.  The module certainly looks like a clone.  WIth any luck though, I might be able to feedback the issue to URUAV and tell them the fix  - they know they have a problem

kilrah, 2019-09-30:
What radio are you using?

CurryKitten, 2019-09-30:
@kilrah I'm mostly using a FrSky X9-Lite Pro which I've had for about 2 months, and an X-Lite (which is a year and a bit old)

kilrah, 2019-09-30:
OK, the X9 Lite could have the PMU "issue" which shuts the module off after a few seconds, but the x-lite should be ok.

CurryKitten, 2019-09-30:
The X-lite I have seemed ok, I had the module on for about 5 minutes and flew a couple of things around - but I'll retest all once I get the bootloader flashed

CurryKitten, 2019-10-01:
@pascallanger I flashed the firmware via ST-flash (happily there was a command line OSX version which saved me searching around for a windows machine) and yes, the new bootloader absolutely fixed the problem.  I've power cycled 20 times and the module always comes up perfectly.

Looking at the instruction from the Vantec module, the board is identical.  I'll feedback to URUAV that there's a problem here (they actually did know something was up as they suggested a workaround) but I'm guessing all these companies buy in the boards and slap there own cases on them, so I don't know if they'll fix it at source... I'd hope they would, else it's a bit useless

pascallanger, 2019-10-01:
Great that it's fixed.
Do you mind to share your email contact for URUAV?

CurryKitten, 2019-10-01:
I don't have one directly, I've having to go through Banggood as a weird sort or intermediary.  Let me see what I can do though - it seems crazy they don't include you in the loop when they are bringing out a module like this.

pascallanger, 2019-10-01:
Ok thanks. I'm closing this issue for now at it is solved but you still can reply to it if you obtain some informations.

closed 2019-10-01 by pascallanger

gretel, 2019-11-09:
@pascallanger one more questions: the bootloader is contained in the `.bin` files i use to flash via `usb` so the difference is flashing via serial instead of usb? thanks a lot

pascallanger, 2019-11-09:
The firmware bin file does not contain the bootloader. The bootloader is automatically installed when a serial flash is detected.

gretel, 2019-11-09:
@pascallanger thank you!

gretel, 2019-11-10:
the guide was very helpful cause took me some time to connect usb, too ;)
did powercycle couple of times.. seems reliable now. thanks again!