pascallanger/DIY-Multiprotocol-TX-Module #277: Create Support for Spektrum DSMR protocol

shorty707, 2019-10-02:
Spektrum surface receivers use DSMR Variant of DSM protocol and cannot bind with 4in1 atm

pascallanger, 2019-10-02:
Are you able to provide on the TX SPI dumps or mail a TX+RX so I can have a look?

shorty707, 2019-10-03:
Don't really know how to do that.
Any guide how to do that?

Here is a Spektrum person telling a little about the background with 5.5ms resolution etc
https://www.rcgroups.com/forums/showthread.php?3003835-Spektrum-air-and-surface-compatibility

pascallanger, 2019-10-03:
Your RCG link is not helping as there is no technical details about the protocol.
You need to solder a couple of wires (6 or so) in your DSMR transmitter and connect to it a logic anlyzer like this one https://www.ebay.fr/itm/USB-Logic-Analyzer-Device-Set-USB-Cable-24MHz-8-Channel-24MHz-for-Saleae-ARM/132431128540?hash=item1ed582a7dc:g:aqcAAOSwywRaLo3e.
Then from your PC you need to do a bunch of dumps to see if we can figure out how the protocol works.
You'll surely have to do the same on the RX side (soldering wires) so we can see the hopping frequencies based on the TX ID.

closed 2019-10-08 by pascallanger