pascallanger/DIY-Multiprotocol-TX-Module #282: Enable a check for _MyConfig.h by default

MRC3742, 2019-10-11:
Would it be possible to add a way to check for the _MyConfig.h by default.
I currently use this option and remove the "//" in _Config.h to enable it.

The problem is every time changes are made to the _Config.h master branch file in order to use "git pull" to update my local files on my computer, I have to go back in and add "//" back, save, then remove them after you pull the new updates. (or delete _Config.h file)
Currently setting it as a default will cause an error compiling if the _MyConfig.h file has not been added to the build.

I know you're busy currently with adding many new features but is this possible?
Or can I doing something different in git to be able to update my local files?

Thanks - Rich

pascallanger, 2019-10-11:
I've already looked at it in the paste. It works on AVR but not STM32 or the opposite I can't really remember. It's a compilator issue.
One way as you suggest is to use git. Do a fork of the code, make your changes to enable my config plus other stuff and commit it. After when you want to upgrade, you just need to say merge with upstream, the new code will appear while keeping your modified config.
You can use GitHub desktop to do the commits and merge.

benlye, 2019-10-11:
We used this syntax in the past:
```
//Personal config file
#if __has_include("_MyConfig.h") || defined(USE_MY_CONFIG)
#include "_MyConfig.h"
#endif
```
It only worked on AVR because the AVR compiler supports `__has_include` but the STM32 compiler does not.

Which makes sense, because it was added to the C++ standard in 2015 and the STM32 compiler we use is from 2014.  

If I recall correctly, the version of the STM32 core we use locks us to that compiler, but I may be wrong about that.  It might be possible to change, but given the headache it caused last time...

pascallanger, 2019-10-11:
I agree, this feature request doesn't warrant a compilator upgrade...
You have the GitHub solution.
You can also make a simple batch file which changes the flag if you want. By the way this is what I have doing the opposite since I'm commenting USE_MY_CONFIG before I push my new code on github.

MRC3742, 2019-10-11:
OK, I understand, and am familiar with, and use Github desktop to stage the changes, commit them and then push them to my fork before creating a PR.

If I make the change in my local files on my computer and commit but not push them, will my local _Config.h download any new changes from your master but keep my local changes?

Just a thought, to simplify things can you just remove the "//" as the default in _Config.h and add a blank _MyConfig.h file to the master build. This _MyConfig.h file would have to be excluded from being compared on a future pull (if that can be done). Because as in my case it would no longer be empty?
 
Some wording in the build instructions would need to be updated but I would be willing to research and submit those changes if this would be a viable option.

I have tested it and it will build successfully with a blank _MyConfig.h file.

As always do what you feel is best for all involved - Rich

pascallanger, 2019-10-11:
I haven't done it this way since there is a big chance that people will override the file by mistake...
@benlye Can we have a pre-processor prog which could look for the file and if it exists then add the build parameter like the others?
Pascal

MRC3742, 2019-10-11:
I also agree with "this feature request doesn't warrant a compilator upgrade..."
but...
I don't understand what you mean by "a big chance that people will override the file by mistake..."

Seems like almost all newcomers will now be using the precompiled versions?
And hopefully the manufacturers too, so that the bootloader will also get installed. - Rich

closed 2019-10-16 by pascallanger