pascallanger/DIY-Multiprotocol-TX-Module #285: Cannot BIND jumper 4n1 to Frsky D/V mode receivers

JRMSR, 2019-10-13:
Have latest version of Jumper 4n1 FW installed, latest version of ersky9xRPro FW installed (221)
can BIND DMS receivers, but cannot BIND any Frsky receivers (VD5m, 
D6FR, V8FR)
Have tried FrskyD, FrskyX, FrskyV         any help???

pascallanger, 2019-10-13:
Have you tried various values of option/freq tuning as suggested in the troubleshooting section: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md
Usually trying with -40,0,40 is enough to get a bind and then you need to tune the value.
Pascal

JRMSR, 2019-10-13:
Wow thanks for the quick excellent reply.  Cannot believe you took the time to do this

Everything worked, and I have all 4 models with different D and V receivers working, EXCELLENT

Cannot thank you enough, 

Jim Morrow Sr

pascallanger, 2019-10-13:
You can always visit the donation page if you are feeling like it: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Donations.md
Pascal

closed 2019-10-13 by pascallanger

JRMSR, 2019-10-13:
will do

JRMSR, 2019-10-13:
I have one more question if possible

would you know if you can have three different Radios (Turnigy 9XR PRO, Taranis X9DP, Taranis Q7XS)  all with three different Jumper 4N1 MultiModules,  BIND & programmed for the same Receivers/Airplanes.
And use them interchangeable, without RE-Bind.

Closed #285 [https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/285].

pascallanger, 2019-10-13:
Yes you can but it's also kind of dangerous so you must understand what you are doing.
If you are the only one using the 3 radios it's ok but otherwise you need to understand that only one of the radio can be used at a time when using the same protocol.
If you want to move forward anyway, you need to compile the code by yourself and in the _config.h file change the line with FORCE_GLOBAL_ID whic h is explained there.

JRMSR, 2019-10-13:
thanks again, sorry to keep asking

If I only use one Radio at a time, do I have to recompile the FW
I do not have Global Variables turned on in any of my EEPROM models, does that matter 

TIA
Jim Morrow
On 10/13/2019 1:21:39 PM, James Morrow <jamesrmorrowsr@centurylink.net> wrote:
I have one more question if possible

would you know if you can have three different Radios (Turnigy 9XR PRO, Taranis X9DP, Taranis Q7XS)  all with three different Jumper 4N1 MultiModules,  BIND & programmed for the same Receivers/Airplanes.
And use them interchangeable, without RE-Bind.

JRMSR, 2019-10-13:
also,what is the RXnum   for in protocol,   can you change this to identify specific AIRPLANES????

pascallanger, 2019-10-13:
"If I only use one Radio at a time, do I have to recompile the FW"
Yes since the firmware is the one containing the information linking the RX to a module. The same FW should be used on all your modules.
"hat is the RXnum   for in protocol,   can you change this to identify specific AIRPLANES????"
You get it, it's for model match.
I invite you to ask your questions on RCGroups instead as other people can benefit from them or answer your questions: https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page1031
For example your topic has been discussed in length and you can learn from it: https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page1014

JRMSR, 2019-10-13:
thanks allot, will do