pascallanger/DIY-Multiprotocol-TX-Module #287: DSM endpoints

boffaa, 2019-10-16:
My setup is Taranis X9D with opentx v2.3.1 and the Jumper 4in1 running v1.2.1.85.

I'm reading that the DSM enpoints are supposed to be 1100 to 1900 at 100% throw and 1000 to 2000 at 125% throw. However in betaflight, I'm seeing 1156 to 1843 at 100% throw and can only get down to 1070 with 125% throw. Higher than 125% makes no changes in output.

pascallanger, 2019-10-16:
Which RX are you using?
I've just verified, on my Genuine DSM RX I do have 1100 .. 1500 .. 1900µs:
[image]
[image]

boffaa, 2019-10-16:
I've tried on 3 so far:
Spektrum SPM4648 - DSMX FPV Racing Serial Receiver
Spektrum SPM4649T - DSMX Quad Race Serial Receiver
An unknown DSMX compatible rx that cam with a bind n'fly.

All were exactly the same values. What rx are you using?

pascallanger, 2019-10-17:
I have tested the 3 DSM RX I have this morning on the throttle PWM ouput with -100% 0% +100%:
 - Spektrum AR6200: 1100-1500-1900
 - Redcon 6 channels: 1100-1500-1900
 - ORX R720X: 1045-1505-1965
None of them have a SBUS output so I can't see what the SBUS output looks like.

I haven't touched Betaflight for a long long time but I'm wondering if there could be a setting somewhere which is applying a ratio on the channels. From the screenshots I've found, you can set the endpoints https://www.atoc2tech.fr/wp-content/uploads/2017/09/betaflight-beecore-radio-config.jpg so you should be ok?
Otherwise I've found this article which even enables to set the range per channel: https://www.droningon.co/2017/03/23/transmitter-betaflight-cleanflight-endpoint-midpoint-mismatch-rxrange/
If you still want to reach the 1000-2000 on your RXs then enable DSM_MAX_THROW in _config.h.

Pascal

boffaa, 2019-10-17:
Thanks for looking into this Pascal,

I'm pretty sure one of those solutions will be fine. I have tried setting
rxrange in betaflight for now and will see how it goes.

My other concern was regarding the airplanes that I'm flying. Originally, I
was using a Spektrum transmitter and when I switched over to the Taranis
with the 4in1 I noticed that I had more throw on the control surfaces.
Since updating the 4in1 firmware, It seems to be closer to what it should
be, but I'm not quite sure if I have enough throw. I can always adjust the
endpoints in the radio, but the reason I opened this issue is because I
thought maybe it needed to be tweaked. Since we are not getting consistent
results, perhaps that needs to be understood first. I will look around more
to see if there is something else interfering.

Adam

pascallanger, 2019-10-17:
I'll close the issue for now but let me know your findings.
Pascal

closed 2019-10-17 by pascallanger