pascallanger/DIY-Multiprotocol-TX-Module #288: Building 1.3.017, SLT protocol Q100 subprotocol has roll and pitch channels reversed

MRC3742, 2019-10-17:
Building using SPort_Send branch, version 1.3.0.17 
On at least the SLT Protocol Q100 Subprotocol, the Roll and Pitch are reversed  (Channels 1 and 2 )

Reverting back to 1.2.1.85 and all works as originally setup.
Is this by design? or is this an issue?
Thanks - Rich

pascallanger, 2019-10-17:
I need a little more info to understand your current config and understand what's going on.
Which channel order is you TX configured for in the mixers?
Which channel order is the module configured for (AETR, TAER....)?
In the MULTI protocol configuration, do you have a line called "Disable Ch. map" or "No Ch. map"? Is it checked?
Pascal

MRC3742, 2019-10-17:
Both Radio (T16) and MPM (Internal or External) are set to AETR.
The Channels are in the right slots, but by reversed I mean on pitch axis, forward becomes backward and on roll axis left becomes right using 1.3.0.17
I am running current builds on OpenTx (compiled 10-17-2019) and MPM 1.3.0.17
I do not see Disable Ch. map? Are you referring to the Radio GUI or the Build of MPM upload?

MRC3742, 2019-10-17:
@pascallanger 
Your fix has corrected the channels that were backwards on Q100.
Also tested today on Q200 and MR 100. These two are also correct.
Closing this issue as it has been repaired - Rich

closed 2019-10-17 by MRC3742