pascallanger/DIY-Multiprotocol-TX-Module #292: Tactic SLT dropouts T-16 internal and less but also in jumper 4-in-1 external T16 & T12

bnrusso, 2019-11-19:
I left notes on rcgroups but no response. 
bench testing Tactic (SLT) with both a Tactic 650 TX
[image]
 and my Jumper T16 on nightlies N34.
PASCAL asked: Can you try instead the latest release 1.3.0.44?
TX and RX about 8 ft apart. I also tried at 20ft apart.I tried 3 different Tactic RXs. And 3 different MPMs. All of the RXs had dropout with T16 N34 w 1.3.0.44 using separately internal and external MPM on SLT.
The dropout was significantly LESS (once per 30-60seconds)with External MPM.
I have 1 RX that seems very sensitive as the Tactic bind activity red light was dim and flickering just as it would act when one reaches the edge of freq tuning. I realize there is no tuning for SLT.
I also tried 2 different name brand Tactic TXs. They both lock on with no dropout to all the RXs. 
If it would help I'd like to get the sensitive RX to whomever may be able to look further into this.
I didn't have to move the sticks, just wait for the red light to pulse off then back on.
I also bound and rebound several times to see if there was a difference. And I tried my T12 with an older 1.2.1.51 and the latest too.
All of the externals had much less problems but it was still there.
I wanted to add a video but it is 50megs.

pascallanger, 2019-11-19:
There is no answer just because there is no solution. In fact, there is a solution: ask Jumper to use better components in their RF board and especially around the NRF24L01 which does not support any tunning. You would see that with the same code an iRangeX module will work without issues.
Pascal

bnrusso, 2019-11-19:
I will try that as I have Irangex too, thank you. I was able to post the video on youtube.https://www.youtube.com/watch?v=DYreU_f1Fug

pascallanger, 2019-11-19:
Ok let me know the result.

bnrusso, 2019-11-21:
You are correct. I flashed my IrangeX first with 1.2.1.85. SLT is solid. Re-flashed with 1.3.0.44 again SLT is solid. Switched off EXT, switched on INT SLT flashes and dims with jumper 4in1. I feel there should be a warning somewhere as I thought it was the programming and not the unit.

closed 2019-11-21 by bnrusso