pascallanger/DIY-Multiprotocol-TX-Module #293: jumper t16 having a low RF signal issue with spektrum dsmx receiver (using JP 4 in 1)

faraz-hub, 2019-11-20:
Facing serious problem when using spektrum dsmx receiver with jumper t16 it binds smoothly but having a low RF signal issue and all signal lost under 15 meters.

pascallanger, 2019-11-20:
If you are sure your that your antenna is well connected (antenna screwed all the way and the tail securily attached on the module in the radio) and that you haven't checked the low power setting under the UI then you have a defective module, contact your reseller.

faraz-hub, 2019-11-25:
Yesterday I use my jumper t16 with dsmx 6ch receiver without telemetry and it worked fine with complete controls and good range check. What I guess it is having only a problem with telemetry receiver.

pascallanger, 2019-11-25:
Ah, I thought you were saying that you were losing connection but it's only telemetry!
Most DSM RX and all in one models have only fly by telemetry unless it's really written on the RX "long range telemetry". You should ignore any telemetry loss signal and even disable the alarms.

faraz-hub, 2019-11-25:
Thanks for your guidance, I have a full range telemetry receiver which
works absolutely fine with spektrum's original transmitter. But with jumper
t16 it disconnects the receiver(receiver switched off) as I walk away and
reconnect when come closer and says telemetry data loss and telemetry
recovered respectively.

benlye, 2019-11-25:
There have been a few reports in RCGroups of people getting significantly reduced DSM range after updating their internal modules from the stock 1.2.1.51 FW to 1.2.1.85.

This, for example, describes the same issue - short range on telemetry-nebled receives only:
https://www.rcgroups.com/forums/showpost.php?p=43185319&postcount=1341

pascallanger, 2019-11-25:
Hummm...
@faraz-hub Would you be able to ship your RX with low range RX so I can troubleshoot?

ChipRey, 2019-11-25:
Ben, I'm totally confused. I installed the internal module and the only protocol choices are D8,D16 and LR12. Is this a module firmware issue? If so, do I need to do the bootloader install mod? Thanks

benlye, 2019-11-25:
> Ben, I'm totally confused. I installed the internal module and the only protocol choices are D8,D16 and LR12. Is this a module firmware issue? If so, do I need to do the bootloader install mod? Thanks

@ChipRey let's take that conversation back to FB and not get this issue (which is about DSM/Spektrum confused.

Rickster250, 2019-11-25:
I am having the same issue, bad RF range with spektrum/telemetry RX's,  all non telemetry dsmx works fine. RF siglnal low and rf signal lost at a fraction of normal range.

pascallanger, 2019-11-26:
Ok so it points out a change between 1.2.1.51 and 1.2.1.85. As far as I recall I've changed the timing for DSM due to the fact that some people were not getting telemetry on some RXs. Let me have a deeper look.
By the way have you tried the latest 1.3.0.44?

pascallanger, 2019-11-26:
This is the part that has changed in DSM_cyrf6936.ino :
```
	#ifdef STM32_BOARD
		#define DSM_WRITE_DELAY		1500		// Time after write to verify write complete
	#else
		#define DSM_WRITE_DELAY		1950		// Time after write to verify write complete
	#endif
```
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/e70bdd4152187ae75c8d24ac0a433f639318d791/Multiprotocol/DSM_cyrf6936.ino#L383
Can you change the value 1500 and try different values up to 1950? steps of 100 should be fine.
The doc to compile yourself the code is here https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md . Let me know if you need help.
Pascal

Rickster250, 2019-11-27:
YES
I had to to use the Opentx daylies.which i am using

pascallanger, 2019-11-27:
And? Are you still having short range with that version?

pascallanger, 2019-11-27:
I've published a new version of Multi v1.3.0.45 with a different timing: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/1.3.0.45
Please test and report.
Pascal

pascallanger, 2019-11-28:
Anyone testing?

Rickster250, 2019-11-29:
@pascallanger  I should be able to get some flying in tomorrow. to try it,

faraz-hub, 2019-11-29:
I will be testing it on Sunday

Rickster250, 2019-12-01:
I flew today with good results, i had no range issues today with a spektrum telemetry sat.
I was going to try going back on the firmware to 44 as a test.
But i clipped a tree branch about 100yards away and it was too late by the time i found my quad. lol
I might be able to fly some more tomorrow.

pascallanger, 2019-12-04:
Not hearing complains anymore so I consider that this is fixed.

faraz-hub, 2019-12-07:
please guide me how to get my transmitter (Jumper t16) ready as I have not
run any kind of update yet so this will be the first one.

pascallanger, 2019-12-07:
@faraz-hub 
check here, they've explained it more than 100 times and there are videos: https://www.rcgroups.com/forums/showthread.php?3243685-Jumper-T16-2-4G-16CH-OpenTX-Multi-protocol-Radio/page486

joshuatreemcd, 2020-02-18:
The short range issue has not be resolved as of yet. I'm using the spektrum 4649T and max range is roughly 30 meters before failsafe occurs. Is there a fix coming?

pascallanger, 2020-02-19:
Do you have another DSM RX where you get full range with your module?
Anyway, unless someone sends me one of the RX with low range, there won't be any fix.

joshuatreemcd, 2020-02-19:
Physically mail you the receiver in question? I can do that.

Rickster250, 2020-02-25:
HI, I wanted to let you know that after the last radio firmware update and module update  V1.3.0.5
I'm getting RF SIGNAL LOW again at a short range, with Spektrum 4649T  Telemetry Satellites.They dont send the the RSSI level to the FC, so I don't have it in a log.I retested with my DX9 TX and i have no issues.
Rick D

Rickster250, 2020-02-25:
V1.3.0.53  i meant

pascallanger, 2020-02-25:
The RF low signal with DSM does not mean anything. The dx9 doesn't complain because it doesn't even display it use this info. Just do the same with OpenTX and disable rssi warnings.
The only thing which matters with DSM is if you get failsafe which is most likely not the case. Even at long distance.
I'm annoyed by people reporting or have just because they get a rssi warning. I'm going to remove it all together.

Rickster250, 2020-02-26:
Hi,  to be honest, im tired of smashing my drone into the ground., When it says RF signal low, i turn it it around and land it.
Disable Telemetry Alarms is TURNED ON
the Warning is not "RSSI LOST"  or low RSSI
Whatever it is, it was fixed until the last radio fw and module fw.
I still have my DX9, 
If i have to use it to fly certain quads, then I will live with it.
Thank you for your work.

pascallanger, 2020-02-26:
Where is this RF signal low coming frrom then?
I have no way to reproduce the issue and therefore solve it.

joshuatreemcd, 2020-02-26:
Rickster250 - Which firmware (radio and module) did it work with?

pasallanger - Can I send you a 4946T receiver to test with?

pascallanger, 2020-02-26:
If you fully explain the reproduction steps of the issue and send the associated hardware that I can just plug and work on, I'm more than interested to get it.
Pascal

Rickster250, 2020-02-27:
I Believe it is coming from the TRSS signal level in the Telemetry screen on the T16.
the spm4649T sends alot of info back to the radio but RSSI is not in the list, for some reason,on the telemetry screen. only TRSS, It has a full scale of 30I do not have the cable to flash my 4649t's to see the RSSI on the FC OSD
RSSI from the 4649T does show on the spektrum DX9 radio, but not on the T16
The Frsky XM+ RX shows TRSS and RSSI
The Frsky R9 mini just has RSSI
If you need anything from me let me know.

pascallanger, 2020-02-28:
So you are saying that you've disabled the RSSI alert but you are turning back when you get an alert. It doesn't make sense to me.
Also I need to make sure that you have tested another type of DSM receiver and you are getting good range with it. Unless you confirm that there is a real difference, I would say that your module is faulty.

Rickster250, 2020-02-29:
Yes That is what I'm saying.I have been flying Spektrum for a long time, most of my RX's and SAT's are spektrum.I have all dsmx receivers in about 15 airplanes, I havent even tried my airplanes with the T16 yet.
I have 8 Quads.2 x spm4649T dsmx telemetry3 x spm4648 dsmx  non-telemtry1 x lemonrx dsmx diversity1 x frsky R-SXM1 x frsky XM+
I fly them all when i go flying.the spm sats all work normal with my Spektrum DX9 radio
I think its pretty obvious what the issue might be.The T16 is not receiving or displaying the RSSI signal,from the spm4649T  all it is tracking is the telemetry signal.

Rickster250, 2020-03-01:
OK so i picked up a cable and flashed my spm4649T RX's.And on the first test, i got no warnings at all, even to fail-safe.It was a ground test, with a gopro pointed at my quad so i could see the RX led later.Still only TRSS on the radio.I will try to do some real flight/range test tomorrow.
RickD
Rickster250

    On Saturday, February 29, 2020, 08:57:56 AM EST, RICHARD DOBERSTEIN <rdobers@bellsouth.net> wrote:  
 
  Yes That is what I'm saying.I have been flying Spektrum for a long time, most of my RX's and SAT's are spektrum.I have all dsmx receivers in about 15 airplanes, I havent even tried my airplanes with the T16 yet.
I have 8 Quads.2 x spm4649T dsmx telemetry3 x spm4648 dsmx  non-telemtry1 x lemonrx dsmx diversity1 x frsky R-SXM1 x frsky XM+
I fly them all when i go flying.the spm sats all work normal with my Spektrum DX9 radio
I think its pretty obvious what the issue might be.The T16 is not receiving or displaying the RSSI signal,from the spm4649T  all it is tracking is the telemetry signal.

pascallanger, 2020-03-02:
You know that you need to discover the sensors when the RX is connected? The fact that you have TRSS displayed shows that there is some telemetry frames being received therefore the radio should show something more than just TRSS... OpenTX only uses TRSS and no other sensors, this is why you need to disable RSSI warnings and set the warnings on whatever sensor you are interrested in.

You still haven't answered my initial question... Have you at least tried to do a range test on one of your receiver like the lemon rx and look at how far you can go without losing the servos? And I say the servos not any warning or anything else coming from the radio. Again the radio can't report any connectivity issue.

Rickster250, 2020-03-02:
All of my drones get flown a lot, and range tested quite often.I get pretty much the same range form the spm4648's and the lemon diversity Sat's
I flew yesterday with the new firmware updated on the SPM4649TAnd i think that the FIRMWARE was the problem. no audible warnings , and RSSI working in the OSD
Everything seems to be normal now, I feel like it should have better range, but i did fly it about 100 yards away.

RickD

pascallanger, 2020-03-21:
Closing this issue since it seems to be in fact due to a RX software upgrade.

closed 2020-03-21 by pascallanger

Rickster250, 2020-04-15:
Hello Sir, I hope you are doing well in this time of Crisis. 
I am having trouble trying to bind  XM+ receivers with the new firmware on the frsky site, to my T16 and it isn't working.multi-stm-opentx-aetr-noinv-v1.3.0.76.bin

XM+_ACCST_2.1.0_RSSI16_FCC.frk         will not bind

XM+FCC170313-RSSI16.frk   will bind

Am i overlooking something or using the wrong firmware?
Appreciate your time.
Rick D

pascallanger, 2020-04-15:
Why are you posting on this DSM issue for something related to Frsky?

Rickster250, 2020-04-16:
sorry,   i used the thread to email you, i thought it could be related to the TX module.But I forgot Frsky broke it with the ASCCT 2.X  firmware updates.
Have a good daystay safeI hope we can get back to normal soon

Rickster250, 2020-05-05:
I wanted to let you know that im having problems again after upgrading to v1.3.0.91  from  v1.3.0.76
I just got back from flying, I had just started to trust the T16 with my 2 spektrum quads with the SPM4649T receivers,
 I bought a spektrum cable so i could upgrade the firmware, The upgrade solved the range problems I was having, until today. 
With the latest jumper module firmware v1.3.0.91, im actually getting fail safes on one of them, and low rssi readings on the other one.
I'm flying in the same places I always fly when I'm trying new stuff. Pretty much open fields.
I re bound my SPM DX9 to my Skyliner with the SPM4649T and my RSSi levels are normal with no fail safe.I get low RSSI levels and fail safes with v1.3.0.91.bin and SPM4649T receiver.
I am also getting reduced RSSI levels with v1.3.0.91.bin used with Frsky XM+  _ACCST_2.1.0_RSSI16_FCC.frk
I just got back form flying with v1.3.0.76.bin and XM+FCC170313-RSSI16.frk, and the RSSI OSD levels are back to normal. I have DVR footage flying the same field with both setups.I also have 1, R-XSR in a quad, but i dont have RSSI in the OSD and ididnt think to turn  audio RSSI on . So I don't know if the R-SXR is affected, Ill have to fly it again.
I don't know what version the problem shows up in. between .76 and. 91.  I will try to see if i can figure it out.
Thanks for all you do.