pascallanger/DIY-Multiprotocol-TX-Module #299: Multistm opentx aetr inv 1.3.0.45

stubugs, 2019-11-27:
T16 with internal multi module works under tools frsky xsr for config x8r 
But when using external module same firmware except inv binds script runs but no data

pascallanger, 2019-11-27:
It works fine. There is something wrong with your setup.
Do you see the version number on your external module in the OpenTX GUI?
Is the internal module off?
Are you getting telemetry?
...

stubugs, 2019-11-27:
I've tried it in my t16 nightly and my x10s the results are the same in the
t16 I get telemtry and int module is off
 on x8r reciever when bound to internal. Module reciever light constent
green and lua xsr works but when on same jumper t16 external module binds
ok but green light flickers fast I get telemetry
But no xsr information the only difference is ones external ones int both
same firmware obviously extcept one inv
Regards stuart

stubugs, 2019-11-27:
Sorry yes version shows in gui
Stu

benlye, 2019-11-27:
Sounds like the external module needs frequency tuning?

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md

closed 2019-12-02 by pascallanger